mriscoc/Ender3V2S1 #1193: [Documentation] Force ABL before each print.

wmantly, 2023-12-31:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

First off, thank you for making this awesome firmware!

I have followed the ABL section of the wiki. Both examples of the gcode to add to the start of a print require an existing mesh level and will not perform the auto level when the print starts. For reasons I can't understand, this is the only documented course. I have been trying to string together the correct gcode to force a mesh level, save and use it at the start of each print.

In this ticket, I am asking 2 things:

1) Please provide the correct gcode (if the below is not) to force a mesh level before each print.
2) Add said gcode to the ABL wiki so others have a choice in how mesh leveling best works for them.

I am sorry to post this here, as this tracker is meant for bugs against the code itself. I fell this is a valid issue with the documentation.

```gcode
G28 ; Home all axes
G29 P1    ; Take a bed leveling mesh.
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M420 S1 Z2 ; Use mesh level upto 2 mm
```

### Version of Professional Firmware

20231202

### Printer model

Ender 3 v2

### Electronics

4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-12-31:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2023-12-31:
Check https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts

closed 2023-12-31 by mriscoc