mriscoc/Ender3V2S1 #1196: Thermal Runaway with REVO CR and Ender3V2-422-BLTUBL-MPC-20231202

ReverendRhyme, 2024-01-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Running the latest firmware. I just upgraded to the Revo CR.

I can do that new PID (MPC?) tuning part just fine, but the moment I set the hotend temp, it climbs well over until I get thermal runaway errors, the printer freaks out and beeps, and I have to shut it off. Reverting to Creality Stock Firmware does not have the same issue. (GD-Ender-3 V2-Marlin2.0.8.2-HW-V4.2.2-SW-V1.0.7_E_N_BLTouch)

### Expected behavior

I exptected the hotend temp to stay at the specified temperature, not overshoot it and keep climbing. Especially after multiple successful PID tunes.expected

### Actual behavior

Set temp for hotend. Temperature overshoots and keeps climbing until a warning is hit.

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231202

### Printer model

Ender 3v2

### Electronics

4.2.2

### Other add-ons

CRTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-01-02:
Ender3V2-422-BLTUBL-MPC uses the T1 thermistor table, try with a custom build with PID enabled.

ReverendRhyme, 2024-03-06:
That was it, MPC is not compatible with REVO CR, and using Special Configurations to build one with PID enabled was the fix. Thank you.