mriscoc/Ender3V2S1 #1197: Thermal runaway stock ender 3v2 with Ender3V2-422-MM-MPC-20231208

notwired, 2024-01-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Was printing a gcode from octoprint installed on RPi and after 30 minutes, tasmota plugin that was monitoring thermal limits reported a thermal runaway and turned off the printer. Octoprint reported that the temperature blinked at 1000 degrees C. 

### Bug Timeline

printed fine with this firmware for many days

### Version of Professional Firmware

Ender3V2-422-MM-MPC-20231208

### Printer model

Creality Ender 3V2 stock

### Electronics

Board 4.2.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-01-02:
Did you removed the +VUSB from the USB cable? https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

notwired, 2024-01-03:
Yes its taped day one. Since yesterday i printed the same part twice and no error. Maybe its the hotend thermistor going bad.

mriscoc, 2024-01-03:
Could be related to https://github.com/mriscoc/Ender3V2S1/issues/889

Fortex365, 2024-01-10:
[image]

4.2.2 + stock e3v2 thermistor and heater, custom build your firmware from latest in december:
[image]

rpi power pin disabled

mriscoc, 2024-01-11:
Your temperature plot is very inconsistent, check your thermistor.

closed 2024-04-11 by github-actions[bot]