mriscoc/Ender3V2S1 #1201: [BUG] Mesh insert wont let you use the whole bed

bauch1234, 2024-01-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

ABL doesn't use the whole bed it and you cant adjust it past y 179 and x 199 also the tramming doesnt go to the back corners 

### Bug Timeline

unknown

### Expected behavior

i expect to be able to adjust the mesh to 220 x 220 

### Actual behavior

it doesnt let me adjust to the edge of the bed 

### Steps to Reproduce

simply try to adjust the points for the mesh 

### Version of Professional Firmware

2.1.3

### Printer model

s1

### Electronics

stm32f401rct6

### LCD/Controller

tcj

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2024-01-05:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

bauch1234, 2024-01-06:
but this is a problem with the software? it was fine when i was running stock software

WutIzGithub, 2024-01-13:
Hello,

Thank you so much for your work on this firmware. I can confirm similar behavior on Ender 3 Max Neo. After flashing the latest pre-compiled v422 with BLT, I am unable to adjust the x axis past 220 or the z access above 250. Normal bounds of the printer are 300x300x320.

Thanks very much again,

mriscoc, 2024-01-14:
Mesh inset is restricted by the physical settings and probe margin defining the leveling area, some work area is physically unreachable due to the maximum positions and offsets. A correct configuration is necessary as mentioned in the [Calibration Guide](https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides) in order to use the maximum leveling area allowed by your printer.

closed 2024-01-14 by mriscoc

bauch1234, 2024-01-14:
in the calibration guide it said it should be configured for an unchanged (stock) printer mine is stock so the config is wrong

mriscoc, 2024-01-14:
> in the calibration guide it said it should be configured for an unchanged (stock) printer mine is stock so the config is wrong

No, you need to do always some calibrations, as z-offset, mesh inset, leveling, MPC. PID, etc.