mriscoc/Ender3V2S1 #1204: Z Offset

TheRa5mus, 2024-01-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

- I installed the New Firmware update on my ender 3 s1 f1 chip. The Z offset has to be set after every pint How I wish I could go back in time to stop me from installing the new firmware.

### Bug Timeline

came with the new installation

### Expected behavior

for the Z offset to stay set at the set position.

### Actual behavior

resets the z offset after every print

### Version of Professional Firmware

Ender3S1-F1-UBL-MPC-20231217.bin

### Printer model

ender 3 s1

### Electronics

f1

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2024-01-06:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-01-06 by mriscoc

bpinkney, 2024-02-05:
@TheRa5mus : 
It sounds like you have to manually "Store Settings" now to write the printer settings and have them persist between boots