mriscoc/Ender3V2S1 #1209: [BUG] Compilation error when disabling LCD_BACKLIGHT_TIMEOUT_MINS

Abasz, 2024-01-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When I do my custom build with the most current (New Year 2024 (20240107)) version I get compilation error when I disable (comment out) the LCD_BACKLIGHT_TIMEOUT_MINS variable:

.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(meshviewerpro.cpp.o): in function 
`MeshViewerPro::drawPoint(unsigned char, unsigned char, float)':
(.text._ZN13MeshViewerPro9drawPointEhhf+0xb8): undefined reference to `MarlinUI::refresh_backlight_timeout()'

EDIT:

Note that I tested the compilation with the default settings included in the Marlin folder (i.e. not with my custom build) and have only the LCD_BACKLIGHT_TIMEOUT_MINS commented out. Compilation still fails with the same error.

### Bug Timeline

Issue starts with 20230805 and occurs for every build. Actually before this version the default was that its commented out.

### Expected behavior

I expect no compilation error

### Actual behavior

The following compilation error

.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(meshviewerpro.cpp.o): in function 
`MeshViewerPro::drawPoint(unsigned char, unsigned char, float)':
(.text._ZN13MeshViewerPro9drawPointEhhf+0xb8): undefined reference to `MarlinUI::refresh_backlight_timeout()'

### Steps to Reproduce

Comment out the LCD_BACKLIGHT_TIMEOUT_MINS in Configuration_adv.h
Start compilation

### Version of Professional Firmware

20240107

### Printer model

ender 3v2

### Electronics

SKR mini

### LCD/Controller

Dwin

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mf7120, 2024-01-09:
I can confirm this used to happen to me when trying to save some space in the F4 chip in the E3S1. So it is indeed a bug.

TheWebMachine, 2024-01-09:
Agreed. Also some [read: my] F4 boards can even lock up if you attempt to turn brightness to 0, like I once tried in start Gcode for Octoprint prints. Within a couple minutes of the brightness getting turned to 0, the board would lock up. I wonder if they are powering some part of the LCD logic off of the backlight power or tied to it somehow, such that when it goes all the way to zero, comms to the board are interrupted. The F4 is definitely finicky about anything involving the LCD.

closed 2024-01-10 by mriscoc

Abasz, 2024-01-11:
I confirm this is working with my config. Thanks for the prompt help.