mriscoc/Ender3V2S1 #1223: [BUG] Creality Ender 3 V2 Neo, remaining time is invalid until near the end of the print

Flybeaux, 2024-01-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When printing is first started, the remaining time shown is a lot lower than expected, but it increases every second
After printing has been going for a while, the remaining time seems to max out at a ridiculous value and then oscillates
During this period, the remaining time seems to bounce up and down
Near the end of the print, the value finally becomes more realistic.
These screen captures were done after 3 hours of print time into a 24 hour print job.
The firmware is showing 241+ hours on the first screen shot and 242+ hours on the second.

### Bug Timeline

Started when I switched from Creality firmware to Professional in December 2023

### Expected behavior

I expect to see accurate remaining time estimates 

### Actual behavior

At one time the remaining time showed 370+ hours remaining, there were actually less than 24 hours remaining.
As I am writing this report, the remaining time has now dropped to 238+ hours, but there are really only 20+ hours remaining.

### Steps to Reproduce

It happens on every print, nothing special to make it happen.

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231217.bin

### Printer model

Creality Ender 3 V2 Neo

### Electronics

4.2.2

### LCD/Controller

Standard Ender

### Other add-ons

Direct drive extruder and Dual Z axis 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[image]
[image]

classicrocker883, 2024-01-20:
I made some fixes here
https://github.com/mriscoc/Ender3V2S1/pull/1182/commits/f2de5883331543f2e0f3a1445bd0854a56cda397

mriscoc, 2024-01-20:
I advice to you to use the slicer method, check the screenshots at the end of: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

closed 2024-04-02 by github-actions[bot]