mriscoc/Ender3V2S1 #1224: [BUG] UBL Mesh grid size resets to 5x5 after restart

Abasz, 2024-01-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have a 7x7 Mesh Grid size in my custom configuration.h for UBL. I attached the complete zip. 

When I flash the firmware its correctly showing 7x7 in the settings menu and UBL starts with 7x7. 

But after I hit "restore defaults" and save, following every restart restart it resets to 5x5. If I click "restore defaults", it goes back to 7x7, but even if I "store settings" (after restore) after reboot it goes back to 5x5.

I tried to fiddle with the settings and the firmware (the order of store, reset, restore or init / no init enable in the firmware ) but this behaviour was consistent. 

No other setting is seemed to be affected by this, meaning that i dont see other settings resetting to a value that has not been set anywhere. 

The only interesting thing I noticed that if there is no valid mesh data (i.e. all values are NaN in the mesh grid). The preset mesh of 7x7 is kept. This issue only comes up when the mesh is reset to zeros. If the Mesh has proper values init or all NaN then it works fine.

**EDIT: This does not happen if I reset the mesh via the "Reset Mesh" point under Mesh leveling setting (though it reset everything to zero) after save and restart the 7x7 grid size is kept.**

Current workaround is that after reset to 5x5, I change the grid manually to 7x7 that reset the mesh to NaN. Then save and then reboot. Then I build the 7x7 mesh and it works.

I tried to look through the code where the override happens, the only unintuitve part I was able to find is that in proui_ex.h  all these gets undefined:

```ccp
#if HAS_MESH
  #undef GRID_MAX_POINTS_X
  #undef GRID_MAX_POINTS_Y
  #undef GRID_MAX_POINTS
  #undef MESH_MIN_X
  #undef MESH_MAX_X
  #undef MESH_MIN_Y
  #undef MESH_MAX_Y
#endif
```

and then reassigned to this:

```cpp
  #define GRID_MAX_POINTS_X meshSet.grid_max_points_x
  #define GRID_MAX_POINTS_Y meshSet.grid_max_points_y
```

But in the settings.cpp file I can see that the assigment is correct:

```
    #if HAS_MESH
      meshSet.grid_max_points_x = DEF_GRID_MAX_POINTS_X; // here vscode intelisense shows these as U7
      meshSet.grid_max_points_y = DEF_GRID_MAX_POINTS_Y;
```

Not to mentione that the code behind has nothing to do with "5" that I get:

```cpp
constexpr uint8_t DEF_GRID_MAX_POINTS_X = TERN(HAS_MESH, GRID_MAX_POINTS_X, 3);
constexpr uint8_t DEF_GRID_MAX_POINTS_Y = TERN(HAS_MESH, GRID_MAX_POINTS_Y, 3);
```

So I suspect that there is a default 5x5 assignment in the code but I was not able to find it.

### Bug Timeline

It started with 2024 new years version

### Expected behavior

The Mesh grid size is kept between reboots and not reseting to 5x5

### Actual behavior

Mesh setting is reset to 5x5 after reboot

### Steps to Reproduce

1. Flash new firmware with mesh grid size set to 7x7 
2. reboot machine
3. check mesh and shows 5x5

But most steps are in the description, so please see that

### Version of Professional Firmware

20240107

### Printer model

ender 3v2

### Electronics

SKR mini

### LCD/Controller

Dwin

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Additional information & file uploads

Full config files: [Ender3V2-SKRME3V3-BLTUBL-T13-IS-LA-MPC-Speaker.zip](https://github.com/mriscoc/Ender3V2S1/files/13988915/Ender3V2-SKRME3V3-BLTUBL-T13-IS-LA-MPC-Speaker.zip)

mriscoc, 2024-01-19:
In UBL, mesh slots store mesh density, Inset and leveling data.

Abasz, 2024-01-19:
> In UBL, mesh slots store mesh density, Inset and leveling data.

I am aware of this. I only use slot zero. Not the other slots.

mriscoc, 2024-01-19:
G29 Ln will load the MxN grid density from a slot regardless what value had the density before. With G29 Sn, you can store multiple meshes in the slots with their own size and density. If you set the mesh to 7x7 and not save valid mesh data, a invalid mesh flag is triggered after reboot. 

So the proper procedure is, set the correct mesh inset and density, execute an auto leveling to fill the mesh with valid data, save the mesh data, additionally you can store settings so the printer will remember what mesh slot was used and load it at power on. That behaviour was introduced since the Anniversary 2023 version to support AML 2.0. If you want the previous behavior, use any version previously to the 20231202 such as: https://github.com/mriscoc/Ender3V2S1/tree/20230904

Abasz, 2024-01-19:
Ok, I think I understand now what is going on. I assumed the "restore default" also cleared the actual Mesh data but I understand now that it only clears the the slot. So after restore I would need to the same procedure as after firmware update. I did not know this latter part as the wiki only talks about the firmware update part where I assumed the EEPROM does not get reset to default .

I understand that it works as intended.

closed 2024-01-19 by Abasz