mriscoc/Ender3V2S1 #1243: [BUG] (support for runout sensor not working)

CalDymos, 2024-02-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have an Ender 3 S1 with filament run-out sensor.
I have activated the sensor in the "Filament Settings Menu" and checked its function via the "end-stop diagnostic page".
If I now start a print and no filament is loaded, the print process starts anyway, i.e. the print process starts without filament.
I have also tried to set the "Runout Active" parameter to LOW, which also did not help.
Changing the "Runout Dist mm" to 1mm doesn't help either.
The run-out sensor remains without function.
Even if I pull the filament out of the sensor during printing, the print does not stop.

Translated with DeepL.com (free version)

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 S1

### Electronics

STM32F401

### LCD/Controller

TJC

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2024-02-11:
The filament runout sensor is only checked when the extruder movement begins, and if the filament is missing, Marlin waits for "Runout Dist" before to launch an alert, of course the firmware only controls that if the print job was started from the SD.

CalDymos, 2024-02-11:
I have also tried this.
When I start a print process from the SD card and no filament is loaded, the print process starts without a message.
I have already set Runout Dist down to 1mm, without success.

In the original Creality firmware, the print stops when the filament runs out and the print process cannot be started if no filament is loaded.

closed 2024-04-22 by github-actions[bot]