mriscoc/Ender3V2S1 #1246: [BUG] display showing weird artifacts

tonysprenk, 2024-02-04:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Since the latest update I’m seeing weird artifacts on the display (see image). Can this be solved by updating the firmware of the display? Or is it more likely to be hardware issue?

### Version of Professional Firmware

20240122

### Printer model

Ender 3v2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[image]

Abasz, 2024-02-09:
I have something similar. It starts after the printer is on for a long time:
[image]
These are random artifacts that start to appear.

I have skr mini e3v3 with the cable mod.

DBNinja, 2024-02-16:
I'd like to also +1 this issue. I only noticed this after the printer was on for several days and after 6+ 3 hour prints (I wasn't paying attention to when this happened). 

Details about my setup:

Ender 3v2 with 4.2.2 board, T5 
Version 2.1.3 MRiscoC
Custom Build with Configurator. Options enabled: BLT - UBL - DWIN - T5 - IS - LA - MPC - Speaker
RE_creality environment. 

[image]

UBL Mesh was 10x10. IS was not changed from default.

All prints via octoprint (print from octoprint).

mriscoc, 2024-02-17:
Duplicate of: https://github.com/mriscoc/Ender3V2S1/issues/749, https://github.com/mriscoc/Ender3V2S1/issues/558, https://github.com/mriscoc/Ender3V2S1/issues/359, https://github.com/mriscoc/Ender3V2S1/issues/328, etc.

closed 2024-02-26 by mriscoc