mriscoc/Ender3V2S1 #1247: My ender 3 s1 extruder motor is not working

Caverius, 2024-02-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I installed this firmware yesterday and after a couple of prints, sprite stepper motor does ot work at all. As in it doest start running when a print starts and it does work when i move it using the movement feature

### Expected behavior

the extruder to move

### Actual behavior

the extruder doesnt move

### Version of Professional Firmware

Ender3S1-F1-UBL-MPC-20240125

### Printer model

s1 f1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2024-02-06:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-02-06 by mriscoc

Caverius, 2024-02-06:
Hi yeah dont worry about it. It was a mechanical issue(the extruder gear
was out if position due to printing at high speed so the motor was moving
but the gears were not) not a software one.

That being said i did have a useful idea for the software. As it stands the
current software of the ender 3s1, only allows you ti move the extruder is
the nozzle is  heated. Which can be an issue for checking if the motor is
working. Maybe in a future update you could make the extruder move
independently from the nozzle temperature, it could be a useful feature.

Cheers and keep up with the good work

mriscoc, 2024-02-07:
Hi, you can move the extruder without heating if set the minimum extruder temperature to 0 (Advanced/Filament settings), that is a safety step.

You can also set the minimum extruder temperature using G-code: [M302 S0](https://marlinfw.org/docs/gcode/M302.html)

Caverius, 2024-02-08:
So awesome. Thank you