mriscoc/Ender3V2S1 #1249: [BUG] Printer occasionally restarts after selecting gcode with thumbnail

Spencer-Yoder, 2024-02-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

After turning the printer on when I select a gcode file with a thumbnail in it, the printer will restart. It will keep happening until randomly after many restarts, it will fix its self and start working. While this is happening, it will not restart when trying to load a gcode file that doesn't have a thumbnail. 

I have tried different scenarios of inserting the sd card and then turning on the printer, as well as turning on the printer and then inserting the card.

https://github.com/mriscoc/Ender3V2S1/assets/25213226/05f6ba5c-a2a6-48c6-91c9-90bdb96de8a4

### Bug Timeline

It wasn't an issue in 20230805, but now happens in 20240127

### Expected behavior

When selecting a gcode file that contains a thumbnail, the gcode file is loaded and is able to be printed

### Actual behavior

The printer restarts.

### Steps to Reproduce

1. Turn the printer on
2. Navigate to a gcode file that has a thumbnail, and select it
3. Observe the printer restart 
4. Repeat and eventually it will work

### Version of Professional Firmware

2.1.3 20240127 - 14:17

### Printer model

S1 F4

### Other add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

[gcode file.zip](https://github.com/mriscoc/Ender3V2S1/files/14173335/gcode.file.zip)

I download the firmware from the Special Configurations repo under the Linear Advance release. ` Ender3S1-F4-LA-20240127.bin`
https://github.com/mriscoc/Special_Configurations/releases/tag/LinearAdvance

Witska, 2024-02-08:
I have the same problem, firmware Ender3S1-F4-DACAI-T13-LA

mriscoc, 2024-02-11:
Thanks for the report, how many attempts approx. are necessary to trigger the bug with that g-code?

Witska, 2024-02-11:
I get the error intermittently. It doesn't matter which g-code, it's the same with all of them. Taking out the sd card and inserting it doesn't help, only turning off the printer for more than 20s.

SYoder1, 2024-02-11:
> Thanks for the report, how many attempts approx. are necessary to trigger the bug with that g-code?

It normally happens right when turning on the printer, and then will continue for ~15 more times. Then just fix it's self and load.

Spencer-Yoder, 2024-02-11:
> Thanks for the report, how many attempts approx. are necessary to trigger the bug with that g-code?

It normally happens right when turning on the printer, and then will continue for ~15 more times. Then just fix it's self and load.

mriscoc, 2024-02-14:
I tried the same G-code with a 4.2.2 board (at least 20 times) without issue, so, it is specific to S1 F4 boards.

classicrocker883, 2024-03-01:
this isn't specific to anything initially. its more prevalent in chips having 256k because it seems the thumbnail image cache isnt being cleared once exited.
for reference:  
https://github.com/classicrocker883/MRiscoCProUI/issues/64 

I was able to fix over on my end.

mriscoc, 2024-03-01:
> this isn't specific to anything initially. its more prevalent in chips having 256k because it seems the thumbnail image cache isnt being cleared once exited. for reference: [classicrocker883#64](https://github.com/classicrocker883/MRiscoCProUI/issues/64)
> 
> I was able to fix over on my end.

That's a different issue related only to the SDcard management, in the official ProUI the image is read in little chunks so the memory is never filled.

Spencer-Yoder, 2024-04-09:
Sorry I am just getting back to this.

I tried with a few different SD cards (the card that came with the printer, SanDisk, PNY) and they all have the same problem. I have noticed that if you change the filament before trying to open a file, it works on the first attempt. I don't know if it has something to do with just waiting, or if a different operation cleans something up.

mriscoc, 2024-04-10:
> Sorry I am just getting back to this.

Check that you have the latest firmware version: Control / Info / Build date

Spencer-Yoder, 2024-04-10:
> > Sorry I am just getting back to this.
> 
> Check that you have the latest firmware version: Control / Info / Build date

I am still running `2.1.3 20240127 - 14:17`. I got the firmware from the Linear Advance release https://github.com/mriscoc/Special_Configurations/releases/tag/LinearAdvance

Spencer-Yoder, 2024-06-10:
I am still running having this issue

closed 2024-08-21 by github-actions[bot]