mriscoc/Ender3V2S1 #1254: No led control menu newest firmware

heidepiek, 2024-02-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

For years now using this firmware. But i changed some things on my first printer and thought install the newest version. No matter what i try, i can`t get the led control feature on the screen. What am i doing wrong? I used the configurator and the newest branche of mriscoc firmware. The leds light up, show startup colors and than the color that is asigned. But i have no control over the leds. No led menu under controls.

 These are my config files: 
[firmware-20240211-024501.zip](https://github.com/mriscoc/Ender3V2S1/files/14231027/firmware-20240211-024501.zip)

And this is the bin file: 
[config files.zip](https://github.com/mriscoc/Ender3V2S1/files/14231025/config.files.zip)

My second ender v2 that runs the austral spring edition, never has or had any issues.: 

https://github.com/mriscoc/Ender3V2S1/assets/104028471/84a999f2-87e5-4203-b25f-7837063cb0f3

### Bug Timeline

new issue

### Expected behavior

Led control to control the leds

### Actual behavior

No led menu, no control over the leds

### Version of Professional Firmware

version 2.1.3

### Printer model

Ender 3 v2 

### Electronics

4.2.7

### LCD/Controller

 dwin lcd

### Other add-ons

sprite extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-02-11:
Maybe some define labels that control that feature were changed in Marlin, I need to check but I don't have the hardware.

heidepiek, 2024-02-11:
Mr. Castillo, may I thank you kindly for this quick response! Appreciated. Tried until 3am last night to get the lights working again with the new firmware. No chance! With the austral spring success! But after your update this morning, voilla it works. Even have an additional menu now! If i use these 3 definitions, i have a case light option and a led control option.
[image]
[image]
[image]
And if I turn the "#define CASE_LIGHT_ENABLE // For use with NeoPixel" off. Then just like my familiar ender 3v2 blue sprite I only have 1 led control menu. 

Now I am not a programmer, but I think I have discovered a small typo. I found that because I turned off the (case light enable). I then got an error message in platformio

[image]

After changing it:

[image]

The led control option worked.

Thanks Miguel Risco 👍 💪 
Here is one happy camper!

Regards Harrie

closed 2024-02-12 by heidepiek