mriscoc/Ender3V2S1 #1256: [BUG] M600 resumes mid air Ender 3V2

Vic152, 2024-02-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

M600 resumes print mid air.

### Expected behavior

When print resumes after M600 - filament change - nozzle is at the height that allows the new filament to stick to the previous.

### Actual behavior

When print resumes after M600 - filament change - nozzle is above the previous print. It prints in the air. to fix I have to lower Z from -1.5 to -1.95

### Steps to Reproduce

Slice with layer change using M600

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20240125.bin

### Printer model

Ender 3v2

### Electronics

4.2.2

### LCD/Controller

Stock

### Other add-ons

CR touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Additional information & file uploads

[xyzCalibration_cube_13m_0.20mm_215C_PLA_ENDER3V2.gcode.txt](https://github.com/mriscoc/Ender3V2S1/files/14319928/xyzCalibration_cube_13m_0.20mm_215C_PLA_ENDER3V2.gcode.txt)

Vic152, 2024-02-17:
[image]

[image]

[image]
 
I in fact did not change the colour but just to illustrate the point.

Vic152, 2024-02-17:
I found workaround by doing this

[image]

PrintGoodCodeBad, 2024-03-23:
Vic152, thanks for this bug report!

mriscoc, thank you so much for your contribution to the 3D printing community! I hope to contribute something of this magnitude someday.

I have experienced a similar issue. I have an Ender 3 V2 with an Ender Extender 400XL kit (400 x 400 x 450) installed. I have a 4.2.2 control board but I will have to double check when I'm able to do so. I have been using Cura 5.2.2. I am using a post-processing script to add in the M600 filament change code at a certain layer height, but I do not currently know how to determine post processing script versions.
The filament change behaves as expected until the print resumes, at which point the tool head appears to return to the print at too high / tall of a height. I initially thought it was possibly a retraction issue related to the much longer bowden tube for the extender kit, so I printed a and installed a direct drive modification, but I still have the same issue. I have a second stock Ender 3 V2 with the jyers firmware. It runs basically the same slicer settings, with the same Cura version, and same computer, and this stock printer with jyers firmware behaves as expected.
I did check the G-code text, and it appears to have the same Z height command before and after the M600 filament change. So at this point, I suspect that the firmware could be handling the command in an unexpected way.

I found the following Reddit post where this issue and firmware was mentioned:
https://www.reddit.com/r/Ender3v2Firmware/comments/yhi4wa/filament_change_effecting_z_offset/

For now, I reverted to Jyers and the print did not have the same issue. I would try and suggest a fix myself, but I’m not really a good coder…yet.

Vic152, 2024-03-24:
My final G code was this. I found myself doing E step calibration, flow rate calibration, and Z offset calibration (after the previous two)

```
G1 E-100 F1500
G92 E0
M600
G1 E2.5
```

Vic152, 2024-03-26:
I think we can close this. It's good to have some detailed record for community.

closed 2024-03-26 by Vic152