mriscoc/Ender3V2S1 #1257: [BUG] Cleaning/prime the nozzle off the table

ms-daniel, 2024-02-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

the printer is cleaning the nozzle practically off the table. See images.

On Orca Slicer:
the printer is cleaning the nozzle practically off the table
[image]

on the printer:
See "Actual behavior"

### Bug Timeline

i installed this firmware for the first time, so in the first print this happened

### Expected behavior

Before installation, cleaning was done on this blue line:
[image]

### Actual behavior

On the printer (red circle):
[image]

### Steps to Reproduce

I just did the usual, I generated the gcode and when I started printing it cleaned the nozzle outside the table

### Version of Professional Firmware

2.1.3 (20240127)

### Printer model

Ender 3V2 Neo

### Electronics

Board 4.2.2

### LCD/Controller

TJC (i did the update)

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

ms-daniel, 2024-02-18:
more images:
[image]
[image]
[image]

mriscoc, 2024-02-19:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware manual [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-02-19 by mriscoc

ms-daniel, 2024-02-19:
> Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware manual [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the [...]

Thanks, and sorry me 🥹