mriscoc/Ender3V2S1 #1258: [f4 chip error] (f4 chip in ender 3 s1 is not flashing onto ender 3 s1 when put into printer)

superior-julz, 2024-02-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Firmware is not flashing onto the printer (modded ender 3 s1) so I cannot use it. I've read tones of things about the professional firmware and how good it is so I would really like it.

### Bug Timeline

new issue

### Expected behavior

I expected to flash

### Actual behavior

Did not flash

### Steps to Reproduce

1. flash .bin file onto blank sd card with less that 8gb of memory
2. put in 3d printer while turned off
3. turn on
4. should flash, did not

### Version of Professional Firmware

Newest

### Printer model

s1

### Electronics

not sure (ender 3 s1)

### LCD/Controller

N/A?

### Other add-ons

upgraded extruder, upgraded build plate, upgraded nozzle

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

mriscoc, 2024-02-20:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-02-20 by mriscoc

superior-julz, 2024-04-13:
@ddproxy  I did this a while ago when I realized the issue, but this worked great. Thank you.