mriscoc/Ender3V2S1 #1260: [BUG] Ender3S1 Reboots and loses setting when SD card is inserted

WeededMedeusa, 2024-02-21:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

As title states, when SD card is inserted in S1, machine reboots and all settings are lost

### Bug Timeline

Last update, not blatest

### Expected behavior

Not reboot

### Actual behavior

Rebooted

### Steps to Reproduce

Insert SD card

### Version of Professional Firmware

20240122

### Printer model

S1

### Electronics

Stock

### LCD/Controller

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

Vitorleoron, 2024-02-24:
I have this problem here too, but mine is an Ender 3 V2, when I turn off the printer and power on the bed mesh settings reset to the default. Other settings are stored as Z-offset!

mriscoc, 2024-02-26:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/1185

closed 2024-02-26 by mriscoc