mriscoc/Ender3V2S1 #1263: Firmware Randomly Stops Printing

Nebtek0, 2024-02-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I walked away from my printer. The settings were fine. I came back and found the printer nozzle stopped in the exact place it was printing from. The screen was the menu page. I know I did not stop the print because if I did, the nozzle would not have left off in the exact place it was extruding when the error occurred. The Firmware literally just ended my print halfway through.

### Bug Timeline

New to me!

### Expected behavior

I expected the print to finish. Settings were fine.

### Actual behavior

Firmware just stopped printing randomly. The nozzle just stopped moving and left off on some random spot on the model. The screen went to the menu page.

### Steps to Reproduce

1. With newest firmware, print something
2. Prepare to be sad.

### Version of Professional Firmware

First release of 2024

### Printer model

Ender 3 S1

### Electronics

No extra electronics, F4 board

### LCD/Controller

TJC

### Other add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

I have attached a picture of exactly what I came back to. Nozzle still on part of print where it was when the issue occurred. The screen was on the menu page. All of the temps have been cooled down. It was essentially like the Printer had called the finish code too soon, yet I have printed this file 4 other times on this printer, with no changes having been made to it. Since it's something digital, it has to have been this firmware.
[image]

mriscoc, 2024-02-26:
i, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

ms-daniel, 2024-03-03:
try using another SD card or formatting yours

closed 2024-05-13 by github-actions[bot]