mriscoc/Ender3V2S1 #1274: [BUG] build fails with missing binary operator before token

baldpope, 2024-03-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Following the instructions laid out here: https://www.reddit.com/r/ender3/comments/10sx1lr/mriscoc_marlin_professional_firmware_for_ender_3/

When I attempt to build, platformio throws errors 
`$ platformio run -e STM32F103RE_creality
Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\MarlinConfigPre.h:39,
                 from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:302:11: error: missing binary operator before token "(" 
  302 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:960:11: error: missing binary operator before token "(" 
  960 | #if EITHER(MORGAN_SCARA, MP_SCARA)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:1761:11: error: missing binary operator before token "("
 1761 | #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:1957:11: error: missing binary operator before token "("
 1957 | #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
      |           ^
In file included from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\MarlinConfigPre.h:39,
                 from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:3011:11: error: missing binary operator before token "("
 3011 | #if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration.h:3309:11: error: missing binary operator before token "("
 3309 | #if EITHER(RGB_LED, RGBW_LED)
      |           ^
In file included from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\MarlinConfigPre.h:56,
                 from c:\users\baldpope\marlin\ender3v2s1\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:310:11: error: missing binary operator before token "("
  310 |   #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1087:11: error: missing binary operator before token "("
 1087 | #if EITHER(INPUT_SHAPING_X, INPUT_SHAPING_Y)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1321:11: error: missing binary operator before token "("
 1321 | #if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1379:28: error: missing binary operator before token "("
 1379 | #if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
      |                            ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1434:11: error: missing binary operator before token "("
 1434 | #if EITHER(HAS_DISPLAY, DWIN_LCD_PROUI)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1492:11: error: missing binary operator before token "("
 1492 |   #if BOTH(M73_REPORT, SDSUPPORT)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:1498:26: error: missing binary operator before token "("
 1498 | #if HAS_DISPLAY && EITHER(SDSUPPORT, SET_PROGRESS_MANUALLY)
      |                          ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:2147:11: error: missing binary operator before token "("
 2147 | #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:2182:11: error: missing binary operator before token "("
 2182 | #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:2190:9: error: missing binary operator before token "("
 2190 | #if BOTH(AUTO_BED_LEVELING_UBL, EEPROM_SETTINGS)
      |         ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:2302:11: error: missing binary operator before token "("
 2302 | #if EITHER(ARC_SUPPORT, BEZIER_CURVE_SUPPORT)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:2388:9: error: missing binary operator before token "("
 2388 | #if BOTH(SDSUPPORT, DIRECT_STEPPING)
      |         ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:3326:11: error: missing binary operator before token "("
 3326 | #if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
      |           ^
c:\users\baldpope\marlin\ender3v2s1\marlin\configuration_adv.h:4074:11: error: missing binary operator before token "("
 4074 | #if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
      |           ^
Error: Failed to parse Marlin features. See previous error messages.
================================================================================================== [FAILED] Took 1.60 seconds ==================================================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:01.603
============================================================================================= 1 failed, 0 succeeded in 00:00:01.603 =============================================================================================`

### Expected behavior

I expected the firmware to compile.

### Actual behavior

Firmware compile fails.

### Steps to Reproduce

instructions detailed here: https://www.reddit.com/r/ender3/comments/10sx1lr/mriscoc_marlin_professional_firmware_for_ender_3/ 

with current branches pulled from github.

### Version of Professional Firmware

prior to flashing

### Printer model

Ender 3 Max Neo

### Electronics

Board 4.2.2 (afaik)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

heidepiek, 2024-03-17:
I see the post you link to is a year ago. In the meantime, the guys have not sat still and improved the firmware every time. Are you still using an older version of the mriscoc firmware? Also, out of curiosity, just tried to make a bin of an ender3 max neo. With the latest configurator and firmware. Here is the result! 
[image]
Don't know if I checked everything right, but according to the specs I read in the issue you are using a 4.2.2 board and it is for an ender 3 max neo. Am not an expert, but trying to think along a bit. Just my two cents.😜

baldpope, 2024-03-18:
If I understand the process, there are 2 repos I'm supposed to clone from, I've cloned both, and ran the configuration gui from the second, which is where the ender 3 max neo firmware branch is. once that step is completed, I copied the .h files and the platformio.ini over to the first repo on branch Ender3V2S1-Released (which if I'm reading your dir correctly is the same repo/branch) - from there I attempt to use the auto build marlin plugin, which is where it fails.

I'm happy to start from scratch if there's a different build process I should follow? Or if you think I've missed a step, I'll gladly try again if you can help point out the step I missed?

```
~/marlin/Ender3V2S1 (Ender3V2S1-Released)
$ git status
On branch Ender3V2S1-Released
Your branch is up to date with 'origin/Ender3V2S1-Released'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   Marlin/Configuration.h
        modified:   Marlin/Configuration_adv.h
        modified:   Marlin/Version.h
        modified:   platformio.ini

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        configurations/Ender3MaxNeo-422-BLTUBL-MPC/
```

heidepiek, 2024-03-18:
If you download the code from this link: https://github.com/mriscoc/Ender3V2S1
And from this link the special configurations: https://github.com/mriscoc/Special_Configurations
Start the py configurator and choose what you need. The configurator wil make 5 files. Copy those files in the marlin folder that you downloaded from the first link and keep in mind to overwrite the old files. I just deleted the bin i made with platformio and used the auto build marlin to build it again. And again succes. 
[ender 3 max neo.zip](https://github.com/mriscoc/Ender3V2S1/files/14631211/ender.3.max.neo.zip)
[firmware-20240318-055812.zip](https://github.com/mriscoc/Ender3V2S1/files/14631221/firmware-20240318-055812.zip)

baldpope, 2024-03-18:
OK, I think I figured it out, though seems silly so I'm just throwing this out here to confirm...

When I initially cloned the repo for the parent(?) Marlin project, and then installed the ABM and PlatformIO extensions to VSCode, I'm guessing that the extensions set themselves up for the project you have open at the time (I don't recall if I was already in that folder or not).  When I moved over to the the Ender3V2S1-Released project (this repo) and attempted to re-use the ABM/PlatformIO extensions, it failed - command not found for platformio

The parent comment I made was based on (and I know this more-so now) when I attempted to copy the custom files into the parent Marlin repo and compile - that failed.  Honestly - that's just on me hoping it would work, should not have expected it to

Back to the this repo, only when I removed both extensions entirely and re-added them while having open the Ender project could I get the platformio command to execute and complete.

I was able to compile the firmware based on the original instructions after resetting - this issue can be closed as likely user error.

closed 2024-05-30 by github-actions[bot]