mriscoc/Ender3V2S1 #1279: not able to compile SKR mini3 v3 with BLT UBL AND NP

blackoutlaw, 2024-03-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

i have an SKR mini E3 V3 and i used the special configuration pwy for getting the config files.  the settings i did was ender3v2, SKRME3V3, BLT, UBL, DACAI, T5, Speaker, NP, MPC.  when i tried to compile it errored with the following:

Linking .pio\build\STM32G0B1RE_btt\firmware.elf
c:/users/theon/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::store_mesh(signed char)':
(.text._ZN14MarlinSettings10store_meshEa+0x36): undefined reference to `unified_bed_leveling::set_store_from_mesh(float const (&) [9][9], short (&) [9][9])'
c:/users/theon/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::load_mesh(signed char, void*)':
(.text._ZN14MarlinSettings9load_meshEaPv+0x60): undefined reference to `unified_bed_leveling::set_mesh_from_store(short const (&) [9][9], float (&) [9][9])'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32G0B1RE_btt\firmware.elf] Error 1

also when I go into any of the menus I don't get a led menu or case light menu

can anyone help me with this issue

### Bug Timeline

started for me today as i just started to use your firmware

### Expected behavior

it to compile and also give me a case light menu and led menu

### Actual behavior

Linking .pio\build\STM32G0B1RE_btt\firmware.elf
c:/users/theon/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::store_mesh(signed char)':
(.text._ZN14MarlinSettings10store_meshEa+0x36): undefined reference to `unified_bed_leveling::set_store_from_mesh(float const (&) [9][9], short (&) [9][9])'
c:/users/theon/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/lib/proui\libproui.a(proui_ex.cpp.o): in function `MarlinSettings::load_mesh(signed char, void*)':
(.text._ZN14MarlinSettings9load_meshEaPv+0x60): undefined reference to `unified_bed_leveling::set_mesh_from_store(short const (&) [9][9], float (&) [9][9])'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32G0B1RE_btt\firmware.elf] Error 1

also if i change to abl it compiled fine but that is when i noticed no menu for led or case light

the firm ware version im using is Ender3V2S1-20240122

### Steps to Reproduce

step 1 use professional firmware configurator and get configs for Ender3V2, SKRME3V3, BLT, UBL, DACAI, T5, MPC, NP, Speaker
step two follow instructions to copy the config files to the marlin folder and the platformio.ini
step 3 use VS and build
step 4 it will get to the end and when it is writing the last file (ie the firmware.elf) it fails

### Version of Professional Firmware

Ender3V2S1-20240122

### Printer model

Creality Ender 3V2

### Electronics

SKR mini E3 V3

### LCD/Controller

original V2 LCD

### Other add-ons

volcano hotend

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-03-20:
I tried the Ender3V2-SKRME3V3-BLTUBL-DACAI-T5-MPC-NP configuration and it compiled as expected.

blackoutlaw, 2024-03-20:
Can you try it in core xy mode with y home to max. Or can you send me the config files may be I have corrupt ones. I have tried it a dozen times with trying different configuration from your customizer with no luck
Thanks

Sent from AOL on Android

blackoutlaw, 2024-03-20:
If you can can you tell me what this error is or what I did wrong. I have been trying to get this compiled for a week now and my printer has a faulty firmware on it that is why I need help. Thanks for your support and good firmware.

Sent from AOL on Android

shoebill-dev, 2024-03-27:
I had the same issue today. I made a typo copying Configuration_adv.h over to the Marlin Folder, so it was using the default Configuration_adv.h file instead of the one generated by the Special Config Tool.

closed 2024-06-06 by github-actions[bot]