mriscoc/Ender3V2S1 #1282: [BUG] Try to Auto Mesh Bed level with G29

Gurke258, 2024-03-31:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I cant use G29 in my G-Startcode. Also when I send G28 and G29 in terminal of OctoPrint.. Nothing happens.

I have the Marlin 2.1.3 MRiscoC Ender3V2-427-BLTUBL-MPC for my BL Touch.

In the manual it is written:

Put in the start Gcode script of the slicer (Cura, Simplify, Prusaslicer, Superslicer, etc.) after the G28 the commands:

G28 ; Home all axes
M420 S1 Z2 ; Use mesh level upto 2 mm
Z2 is for set fade height to 2, so leveling correction is active only the first 2mm. S1 enables the leveling system (only if a valid mesh exists in RAM), S0 disables the leveling system, more info in [Marlin Docs](https://marlinfw.org/docs/gcode/M420.html).

Note: if you have your printer connected to Octoprint, and you try to print a g-code with the M420 without a valid mesh you can get an M112 error. Please follow [this guide](https://github.com/mriscoc/Marlin_Ender3v2/wiki/Octoprint#error-handling) to change the error handing in Octoprint.

For UBL you can alternatively use these start G-codes after the G28:

G28 ; Home all axes
G29 L0 ; load a valid mesh from slot 0
G29 A  ; active the UBL system
Verify that you don't have a M501 in your start G-code script, that G-code will clear the UBL mesh data, so if you use M420 S1 to enable leveling you will have a mesh with all points in 0.

But i dont have UBL.. i have BLT ^^ or is BLT not for BLTouch?

### Bug Timeline

last week

### Expected behavior

That the G29 do my auto level mesh

### Actual behavior

nothing when i try to use G29

### Steps to Reproduce

1. G28
2. G29

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-427-BLTUBL-MPC

### Printer model

Ender 3 V2

### Electronics

4.2.7

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Additional information & file uploads

I Use teh Orca Slicer V1.9.1

mriscoc, 2024-04-03:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

Gurke258, 2024-04-04:
That problem is not seems like a normally Support issue. I think the Problem is in the Firmware.

I dont have a UBL. I have a BLT. For me its Diffrent and also for Marlin.

So i want to use G28 and then G29 in my startcode and this doesnt work!

mriscoc, 2024-04-05:
[image]