mriscoc/Ender3V2S1 #1283: "Probe out of bounds" error on bed tramming wizard.

SerbianLastName, 2024-03-31:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When I try to run the bed tramming wizard, it will probe the first point, then throw the out of bounds error.
The weird thing, is that if I make any update to the mesh size, the tramming wizard will work, but then have the same issues after reboot, even though the new bed size would have been saved.  I just end up adjusting mesh size +/- 1 after a reboot and the tramming wizard will work fine, but it would be nice to not have to do that.

### Expected behavior

Running full tramming wizard on all 4 corners.

### Actual behavior

Only probes one point then throws error.

### Steps to Reproduce

Run tramming wizard.

### Version of Professional Firmware

Latest

### Printer model

Ender 3V2 with CR touch.

### Electronics

4.2.2

### LCD/Controller

Stock LCD

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

toddbass10, 2024-04-02:
Have the same issue with an Ender 3 v2 NEO.  Was working fine and now I'm getting this error.

SerbianLastName, 2024-04-02:
Now, on one of my Ender 3s, when doing the Tramming Wizard, it just goes to the back right corner, probes it 4 times, and says my bed is perfectly level.

toddbass10, 2024-04-03:
Hell, the one Ender 3 it wasn't working on now it does.  Oh well.

mriscoc, 2024-04-03:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

SerbianLastName, 2024-04-07:
This is a not a general printer support question.  This is a bug in your firmware.