mriscoc/Ender3V2S1 #1285: [BUG] Screen brightness going down somethimes when increasing value

PilotOfGolf, 2024-04-04:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When changing the screen brightness, the screen brightness when increasing the value  usually goes up, but it goes down at one point. 

### Bug Timeline

It has presisted since I have had the firmware (2-3 months)

### Expected behavior

I expected the screen to increase the brightness as I increased the value. 

### Actual behavior

When I increased the value, the brightness usually increased. However at one point, the brightness decreased when increasing this value. This specifcally happens when going from 4 -> 5. 

### Steps to Reproduce

I was able to get this to reproduce when changing the brightness whilst printing, and whilst not printing. I put the steps to reproduce while not printing
1. Navigate to 
2. 

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 S1

### Electronics

F4 Board (stock board)

### LCD/Controller

Stock E3S1 Screen

### Other add-ons

None related

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

This video shows the issue. https://drive.google.com/file/d/1XuYBCbVtvJy80d7Nw-c6eB_T4JGRJYVM/view?usp=drive_link

mriscoc, 2024-04-05:
Some displays units have that behaviour with the generic firmware version, there is not an unique Stock E3S1 Screen, you must to compare your screen with these 5 possible versions: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display
So, some displays need an custom firmware compiled for that units.

mriscoc, 2024-04-10:
For TJC displays the brightness range is defined like:
```cpp
  #ifdef TJC_DISPLAY
    #define LCD_BRIGHTNESS_MIN 6
    #define LCD_BRIGHTNESS_MAX 40
    #define LCD_BRIGHTNESS_DEFAULT 20
  #else 
```

closed 2024-06-21 by github-actions[bot]