mriscoc/Ender3V2S1 #1291: [BUG] Numbers in auto leveling not shown unless bed tramming wizard is run first (TJC)

floorcat, 2024-04-20:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

Mesh leveling doesn't show numbers unless bed tramming wizard is run first.

### Bug Timeline

After 2023-09-04 version

### Steps to Reproduce

1. Boot printer
2. Run auto bed level
3. Notice there is no numbers shown
4. Run bed tramming wizard
5. Numbers are shown correctly
6. Run auto bed level
7. Numbers are now shown

### Version of Professional Firmware

Github master as of 2024-04-20

### Printer model

Creality Ender S1 F4

### LCD/Controller

TJC

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

I compared how calling of tramming wizard differed to calling of auto leveling in the code and noticed that before (and after) calling `trammingWizard()`, `meshViewer.meshfont` is set to `font8x16`. I couldn't find similar thing for auto leveling. 

I then compared 2023-09-04 version to current github master and saw that `meshViewer.meshfont` was set in `dwinSetDataDefaults()` and `dwinCopySettingsFrom()` functions in the 2023-09-04 version but not in master, so I added it back in them and now auto leveling shows numbers normally.

heidepiek, 2024-04-20:
I also have this problem at the moment with my printer and tjc screen. I had a dwin screen. And recently using aml. But yestrrday my rotary encoder got faulty of my dwin screen. So i had no choice to use my spare tjc screen. Because i use aml, every time hé does a mesh. But always without showing the numbers. Also missing the nice pictograms. Too bad that creality produced these bad screens!  Its always a surprise what screen you get when you order a spare one. I have 3 printers that run professional firmware. 2 have the dacai screens, and one had the dwin. And now unfortenetly a tjc! I ordered a New one, but the last time i ordered a screen i got the tjc. I hope  this time i get the dacai or a dwin. I did not want to open a discusion, but now someone did. I can confirm that hé is not the only one. I hope that in the future some clever programmers can also make the tjc screens function just like the dwin and the dacai screens. I was so content with my ender 3 v2 purple Sprite printer with his nice dwin pink screen and nice pictograms. Now with the tjc screen, i feel a little bummed! But on topic. I too have the problem with the numbers!  I know that maintaining this site and keeping this nice firmware up to date is a time consuming job. Hope that the tech guys find a solution for this visual flaw. But what for functionality goes, the first layer is spot on! Aml does its job! Again thanks Miguel and consorts!  

Greetz harrie

mriscoc, 2024-04-21:
> I also have this problem at the moment with my printer and tjc screen.

I hope that you had read the Wiki and know that there is a firmware compiled specially for TJC displays.

heidepiek, 2024-04-21:
Yes i know miguel , i read the wiki. I compiled my version with your configurator. I had it bedleveling with the numbers. But as i told in my story i had problems with my dwin screen. I soldered a new rotary encoder on the print, so i had to flash the printer again to test the dwin screen. But no luck, so i had to switch back to tjc. But maybe i forgot something in the configurator. Is late now, will try again tomorrow. Report how it goes.

Harrie

mriscoc, 2024-04-21:
Please after boot, reset settings by menu or sending the G-code M502 and check.
Could you please test the precompiled versions for TJC, seems to found a possible bug only evident on custom version.

heidepiek, 2024-04-22:
Am totally confused now. Have reinstalled the firmware. Used the M502 Code. And did a bedmesh. Numbers in the squares. Started a print with aml. Again numbers in the squares. But if I turn the printer off and on again and start another print job the numbers are gone????? Did not use an original bin from tlc because my printer was not listed. Have a 4.2.7 motherboard with a sprite extruder and use la and aml. Did try the configuration files that were in the tlc folder. But to no avail either. I don't get it, what am I doing wrong? The screen also has the firmware on it for the tlc. It also works, only after a fresh install. And 1 time printing. As soon as I restart the printer the numbers are gone!

Harrie

floorcat, 2024-04-23:
> Please confirm that you are compiling your custom version and used the Configurator option for TJC displays ![image](https://private-user-images.githubusercontent.com/2745567/324272397-92f5152b-5ee2-490a-8012-afe105841629.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM4ODEzODAsIm5iZiI6MTcxMzg4MTA4MCwicGF0aCI6Ii8yNzQ1NTY3LzMyNDI3MjM5Ny05MmY1MTUyYi01ZWUyLTQ5MGEtODAxMi1hZmUxMDU4NDE2MjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjNUMTQwNDQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGYyOTcyNmZmYzQ5OWViOTIyMDVlM2M3MDJiNDUyOTgxMTgzNTJhOWQxNzU3ZGI5MGYyNWE2NjNmMjczMTFmOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.-EfCODe9B52DWxC9cMv2e4_G2XaB1I5xbV9KidEqWCo) or you can add:
> 
> ```c++
> #define TJC_DISPLAY
> ```
> 
> in **configuration.h** after:
> 
> https://github.com/mriscoc/Ender3V2S1/blob/0e7a9c1fb16324d0d7df62a58c64f4a374ecb2e4/Marlin/Configuration.h#L3409

Yes I did.

> Please after boot, reset settings by menu or sending the G-code M502 and check. Could you please test the precompiled versions for TJC, seems to found a possible bug only evident on custom version.

Unfortunately I can't test this now since I managed to break my mainboard in an unrelated electrical shorting incident. However I remember trying settings reset from the menu and it had no effect on this bug.

heidepiek, 2024-04-24:
After every restart of the printer you have indeed need to tram the bed to have the numbers visible in the squares with bedleveling. Because i am using aml, everytime i have printed something smal, the pionts are set to 3 by 3. So if i tram, the probe is not over the knobs but in the 3 by 3 space where he printed the last time. Bummer!!

mriscoc, 2024-04-24:
> So if i tram, the probe is not over the knobs but in the 3 by 3 space where he printed the last time. Bummer!!

If you want to return to a full bed tramming add  C29 M C (maximize and center mesh) at the end of the End G-code script.

heidepiek, 2024-04-24:
Wil try that miguel. But what about every time after turning printer off and on again, you have to do a tram to make the numbers visable in bedleveling.  Is there a fix for that?

mriscoc, 2024-04-24:
> Is there a fix for that?

Will be a fix in the next release

closed 2024-07-05 by github-actions[bot]

heidepiek, 2024-09-09:
Miguel, when is there a next release? I still have to do a bedtramming wizard first before i see numbers in my screen when it uses aml. Is there a simple fix for that? Maybe a couple of lines that have to be altered?

Greetz harrie

mriscoc, 2024-09-09:
> Miguel, when is there a next release? I still have to do a bedtramming wizard first before i see numbers in my screen when it uses aml. Is there a simple fix for that? Maybe a couple of lines that have to be altered?
> 
> Greetz harrie

I don't have an ETA, because I haven't had the time to do the test, I posted an experimental version in the Telegram group but I didn't received any reports.

heidepiek, 2024-09-10:
Hi Miguel,

Thank you for your quick response! I wanted to check if I’m one of the few experiencing this issue, especially since there hasn’t been a response in the Telegram group. Unfortunately, I don’t have the technical knowledge to resolve this myself. The person who initially raised the issue mentioned that they fixed it by restoring something from an older version, but from what I can see, those parts seem to already be included in the latest version.

I know you’re busy, but I hope you’ll be able to find some time to address this. For context, the printer works well overall, and the AML is functioning fine, but it’s disappointing that the numbers aren’t showing up in the squares—only the colors are displayed. While this still works, I know it’s possible to show both colors and numbers without having to use the tramming wizard each time I turn on the printer.

Thanks again for your help!

mriscoc, 2024-09-10:
> Hi Miguel,
>
> Thank you for your quick response! I wanted to check if I’m one of the few experiencing this issue, especially since there hasn’t been a response in the Telegram group. Unfortunately, I [...]

You don't see the numbers because maybe you have a TJC display, for that kind of display you need an special build for that screen units, because the TJC doesn't support small fonts: https://github.com/mriscoc/Special_Configurations/releases/tag/tjc

heidepiek, 2024-09-11:
Hi Miguel,

Thank you for your suggestion regarding the TJC display. I have already applied the special build for the TJC screens, so I believe the issue lies elsewhere.

Let me clarify the issue further. When I first power on my printer and start a print job, the AML (Auto Mesh Leveling) runs and the colored squares are displayed, but without any numbers inside. However, if I perform a bed tramming after that, the printer probes the four corners of the bed, and this time it shows both the colors and the numbers inside the squares. After this bed tramming, if I start a new print job, the AML runs again, and this time the numbers do appear inside the colored squares.

The problem only seems to occur on the first bedlevel after turning on the printer. Once I perform a bed tramming, the numbers appear as expected in both tramming and subsequent bedleveling processes. If I turn the printer off and on again, the issue repeats: the first bedlevel shows only the colors, but no numbers.

I have made three videos to demonstrate the issue:

    First print job after powering on (no numbers).

https://github.com/user-attachments/assets/7b437ee7-f87c-4a58-9d0e-364fcc76e58b

    Bed tramming (numbers are shown).

https://github.com/user-attachments/assets/710eabc4-3fbd-4eaf-b65d-93c3670e8d41

    Second print job after bed tramming (numbers are shown).

https://github.com/user-attachments/assets/12e526a3-2148-466c-99d9-a39a80af1eac

I believe this may be a firmware issue, and I hope you can help me resolve it. Thank you again for your time and assistance!

Best regards,
Harrie

heidepiek, 2024-09-17:
This was your last answer to my question miguel:
> 
> You don't see the numbers because maybe you have a TJC display, for that kind of display you need an special build for that screen units, because the TJC doesn't support small fonts: https://github.com/mriscoc/Special_Configurations/releases/tag/tjc

In my last post I told you that was not the problem for me. I have been waiting anxiously for a reply from you for a few days now. I know you are busy and can't tell when that there will be a next release. But after your last reply I got the idea that you thought the problem was because of the tjc screen and I was using the wrong firmware. But as you can see in the videos that is not the case. I can indeed get the digits in the screen, but only after bedtramming. Have already had many mail contact with creality to get a dwin screen. But that seems to be very difficult. Also the open source files of the tjc screen are not released. Even after I insisted that it was ridiculous that this was done for the dwin and dacai screens but not for the tjc. I never had any problems with my old dwin screen! I was hoping for an answer from you with something along the lines of, we are going to try to fix this. Or I know there are more with this problem. But so far nothing, no response. 
Maybe I am too impatient, but I am so bummed that my dwin screen broke and now with that stupid tjc screen I got this problem. Am maybe a bit autistic, but I had my three enders running so nicely with your firmware. And so now 1 of them is ruined in my eyes! I now dare not touch the rotary button with my other 2 printers that are thus functioning properly.  And I actually only use the printers with octoprint, to save the rotary button. Really hope you guys figure something out on this! I'm not the only one experiencing this, am I? 😟 If you find me annoying you should say so miguel! Then I'll stop asking, and leave it at that.

Harrie

mriscoc, 2024-09-17:
Hi, this bug was fixed in the source code and I will publish it the repository, you can compile it, request a compile service in the Patreon site or wait to the next release of free binaries, sadly, I don't have a date for the the last.

heidepiek, 2024-09-18:
Flashed the firmware with the bug fix. Now it is working properly 😆 . Thank you very much miguel!👍  Now to convince creality to release the open source files of the tjc screen.😓   Then maybe the icons can be changed.

Regards 
Harrie