mriscoc/Ender3V2S1 #1293: [Issue] Low bed heater PWM frequency powersupply noise

keever50, 2024-04-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

This is not a bug, but rather an annoyance or an issue.

My Ender 3 v2 printer has a custom power supply to make it more silent and the bed heater is the biggest power consumer.
Now the heater PWM frequency is quite low, i assume 2 or 4hz.
This is audible, because my power supply keeps adjusting for the voltage drop and makes a "vibrating" noise, which is ofcourse that 2-4 hz PWM frequency.

Is it possible to add a feature to adjust this? Or is there a way to adjust this myself by recompiling even?

.

### Expected behavior

As example, in klipper it is possible to increase the PWM frequency to something higher like 100hz or multiple kHz, which fixes this issue. No noise or interference is present.

### Actual behavior

A very audible switching noise around 2-4hz on power supply when bed heater is on, which causes annoying noise and interference.

### Steps to Reproduce

1. Custom power supply
2. Turn on heater bed
3. Wait till it balances out
4. noisy  yes

### Version of Professional Firmware

Ender3v2-422-MM-MPC

### Printer model

Ender 3 v2

### Electronics

4.2.2 board

### LCD/Controller

Stock

### Other add-ons

Custom external power supply (Passive, no fan)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-04-27:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2024-04-27:
Hi, I think that your request is more related to the Marlin core, maybe you can ask in the Marlin forums if that is possible to be implemented.

classicrocker883, 2024-05-01:
you can try using an [EMI filter like this one here ](https://www.amazon.com/Filter-CW2B-10A-T-Socket-Single-Safety/dp/B0867MXFXY?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&psc=1&smid=A160PCGAWPEDXQ)

keever50, 2024-05-01:
Thanks for the suggestions. I thought this repo was a Marlin modification to enable some of its features. It appears i have to make a custom build to enable this feature. However, I haven't figured out which exact one just yet.

keever50, 2024-05-01:
> you can try using an [EMI filter like this one here ](https://www.amazon.com/Filter-CW2B-10A-T-Socket-Single-Safety/dp/B0867MXFXY?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&psc=1&smid=A160PCGAWPEDXQ)

Thank you. My power supply has one built in already. However, the EMI is on the printer itself, not the grid ^^
The "EMI" i am mainly talking about are voltage dips because the low frequency switching. Not something easily solved by adding a capacitor or filter.

closed 2024-07-12 by github-actions[bot]