mriscoc/Ender3V2S1 #1294: [BUG] Tramming Wizard won't go to all 4 corners and Autobuild Mesh is not moving correctly

TimothyGaray, 2024-04-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I'm writing here to maybe get some additional troubleshooting on this particular issue.

I have two Creality Ender 3 Max Neo printers with v4.2.7 silent boards and loaded with the latest professional firmware (Jan 2024).

One printer prints fine and otherwise appears to work fine.  However, when you run the tramming wizard it: homes, does front left, front right, back right then doesn't move position but says it is doing back left.  I have to disable steppers and physically move the head back to the left all the way before I can try another tramming.  If I don't, it starts from there then grinds when it hits the end of the carriage.

Shut down, reboot, re-ran calibrations.  Re-applied firmware, restored settings.  Same thing.  If I click the manual tramming to back left it goes there just fine.  I can move it all over the bed with no issues.  Only the tramming wizard is doing this.

Other identical printer works just fine with same exact settings.

This morning I discovered that it also won't do a mesh auto build.  It homes then starts taking measurements but the Y position doesn't change and the X position only keeps inching across the bed to the right until it hits the end of the carriage.  Doesn't matter what it shows on the screen for the point it is trying to measure.

This has happened on this printer in the past and then after a few tries it would start working normally.  Not sure how it gets working again.

### Bug Timeline

A month ago

### Expected behavior

Tramming wizard: auto home, front left, front right, back right then back left.

Mesh auto build: moves around the board in various locations to take readings.

### Actual behavior

Tramming wizard only does: auto home, front left, front right, back right then doesn't change physical position but takes another reading and calls it "back left".

### Steps to Reproduce

Run bed tramming wizard.

### Version of Professional Firmware

New Year 2024 (20240122)

### Printer model

Creality Ender 3 Max Neo

### Electronics

Creality  v4.2.7

### LCD/Controller

TJC

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I've tried unplugging the OctoPrint Raspberry Pi.

I can still print just fine so not sure why this is happening.  I would think if bad/broken cable then I would have problems no matter what.  Same thing if bad stepper motor - why only show up during bed tramming and mesh creation?

Not sure if there is a log available somewhere that might say what it is doing or thinks it is doing when doing a tramming wizard.

If it helps, next chance I get on it I will record a video or two so you can see what it is doing.

I starting by posting this issue on Reddit but not gotten any replies and I am finding more issues (i.e. auto mesh build).
https://www.reddit.com/r/Ender3v2Firmware/comments/1carl8z/ender_3_max_neo_printer_physical_settings/

TimothyGaray, 2024-04-26:
The only thing that seems to remotely make sense to me might be a bad system board (v4.2.7 silent).  I do have a new one on the way but won't be here until Monday according to tracking.  I think I can reconcile issues on the board presenting this way (maybe bad electronic component, chip or memory).  I would think if there was a broken wire or bad motor that nothing would work but so far I can still print on it just fine.

TimothyGaray, 2024-04-26:
So you increased the bed size and that made it work again?

Koky05, 2024-04-26:
Yes.

mriscoc, 2024-04-27:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

TimothyGaray, 2024-04-27:
Sorry mriscoc.  I subscribed to the patreon and will continue there.

closed 2024-04-27 by TimothyGaray

mriscoc, 2024-04-27:
> https://drive.google.com/file/d/1UMwLYo4TbRjAwN25KAU1wOF-IiUeZZWY/view?usp=sharing

Thanks, If you have a CRTouch, please disable HS mode.

TimothyGaray, 2024-04-27:
I did try that as well after reading about CRTouch having some incompatibility.