mriscoc/Ender3V2S1 #1300: Auto mesh isn't working

whop3r, 2024-05-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Auto mesh is moving randomly and clocking the wrong places

### Bug Timeline

As soon as setup

### Expected behavior

I expected it to auto level

### Actual behavior

it would randomly move to select points, it mostly went to the right of the printer.

### Steps to Reproduce

Install on ender 3 v2 with a cr touch
auto level

### Version of Professional Firmware

20240125

### Printer model

Creality Ender 3V2

### Electronics

4.2.2

### LCD/Controller

original 

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

whop3r, 2024-06-16:
can someone help?
I've been trying to fix it for many, many months.

mriscoc, 2024-06-16:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-09-20 by github-actions[bot]