mriscoc/Ender3V2S1 #1309: Z Wizard not working properly - Final print setting way too high.

Loremwriter, 2024-05-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I use the latest version and flashed it today. Got a problem with the z wizard. Used „prepare“ -> z wizard, and set it to a correct high by using the usual paper method. 

Then I stored the settings in the advanced submenu and started printing. 

Result: The nozzle is way too high - like 0.50 too high. Had to use the „on the fly“ settings to bring it down to the correct height for the filament to stick to the bed. 

Don’t know if I have to select UBL or ABL. I use the standard BL Touch system on the S1 Printer. 

### Bug Timeline

Since today (day one)

### Expected behavior

I expected to run the z wizard and then using this (correct) setting while printing without any „on the fly“ changes. 

### Actual behavior

Nozzle is too high even though the z wizard was used.

### Steps to Reproduce

1. Boot
2. Prepare - z wizard 
3. Print and the nozzle is too high 

### Version of Professional Firmware

Latest version - Will update this line when the printer finished its job. 

### Printer model

Creality Ender 3 S1 - F103

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

lorenmauch, 2024-05-28:
I am also having this issue on my Ender 3 V2. Mainboard version 4.2.3 with CR Touch. Used this prebuilt firmware: Ender3V2-422-BLTUBL-MPC-20240125.bin Thanks for the amazing firmware. I have used several previous versions without any issues.

AllanMartel, 2024-06-12:
I also had this issue with the firmware **Ender3V2-422-BLTUBL-MPC-20240125** on my Ender 3 v2. I would set the Z offset and start a print, only to have to readjust downward the Z height with the TUNE feature immediately. The distance would be pretty far; Z wizard says offset is -2.45, but when you start to print and readjust the Z, I end up at -3.60.

I ended up using the configurator to create a custom firmware after purchasing a Sprite Pro, and I have not seen this issue again.

lorenmauch, 2024-06-13:
I think I fixed my problem on my  Ender 3 V2. Mainboard version 4.2.3 with CR Touch with the Ender3V2-422-BLTUBL-MPC-20240125 firmware. (See my previous comment above.) I recently upgraded to Dual Z and replaced my CR Touch module at the same time. Everything is working fine now. Thanks!

adommett, 2024-08-16:
I also have installed the 4.2.2 with bltouch firmware and am having issues. And the z wizard comes out with large swinging differences - anywhere from -.68 to -3.89. I gouged out a mat before I caught it. I have rotated through making a mesh first and saving that under the advanced-save setting option and the using z-wizard (also saving after each of those steps. And the Z offset never measures the same and often has the print too high or crushed into the mat.

Loremwriter, 2024-08-16:
Is there any way to fix this? 

It seems to be a common problem on these E3S1. Right ?

Loremwriter, 2024-08-16:
Oh.. and it’s a bug. Not false alarm. ;)

Or „are we using it wrong“?

adommett, 2024-08-16:
I'm using an Ender 3 V2 (for reference).

Sch1nken, 2024-09-02:
Same issue on my 4.2.2. Using a custom configured build does not solve the issue. Nozzle is way too high while printing. Distance seems to vary from run to run as well...

Upgraded from some 2022 Version of the Firmware straight to 01/24 one so can't really tell if the issue started somewhere in-between...

BavarianLord29, 2024-10-11:
I have the same problem with Z-wizard. Although I know for a fact that my Z-offset is -1.68 mm, so I go to print the nozzle is always higher than at least 0.50 from the plate. As I do the bed tramming wizard, The values change with each measurement and the bed mesh has crazy readings that were not there with the 2023 version. I changed the firmware on my Ender 3 V2 (4.2.2) to the new one just yesterday and I am feeling about it. I should have read the comments first.

Solutions?

I tried setting the Z-offset on Orca slicer as well, something seems to have changed but it is not accurate, either too close to the bed or too far away. Help!

Loremwriter, 2024-10-11:
Would appreciate any help. Right now I'm using the stock firmware, because it simply works. The Z-Wizard is totally off. 

Any help?

BavarianLord29, 2024-10-11:
Is it not possible to download a working version like 2023??!! I can't find it anywhere!!!

Sch1nken, 2024-10-11:
Iirc someone suggested compiling the latest master themselves. And I think that's what I did. I can't remember, haven't used my printer in a while...

Check https://github.com/mriscoc/Special_Configurations for instructions. It's not too hard if you've got some experience.

Loremwriter, 2024-10-11:
Haven't tried it with compiling the lastest master. Maybe I'll find a solution this weekend. 
But why is the uploaded firmware not working properly. I guess it should be the right one for a usual stock Ender3 S1, right?

It's sad, since I have this problem for more than half a year now..

If anyone compiles a working firmware, please comment! :) 

Sch1nken und BavarianLord29, alle deutsch hier unterwegs? ;)

BavarianLord29, 2024-10-11:
Yup 🫣