mriscoc/Ender3V2S1 #1312: G92 doesn't behave preserve physical limits and physical mesh offsets

lagnat, 2024-06-03:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

- My intention is to use the 3d printer with a drag knife.  Sometimes it's necessary to manually jog the carriage to a specific location before sending the gcode.  To do this, I use the knob screen to get where I want to be, enable the mesh, G92 X0 Y0, and then start the job.  This is not working as expected.

### Steps to Reproduce

These are my observations (steps to reproduce)
* Home X, Y, Z
* Manually jog the carriage some away from 0,0
* Enable the mesh.  Upon doing so, observe the Z value on the knob screen change slightly.  This confirms that the mesh is enabled and working.
* G92 X0 Y0.  My expectation is that the Z value doesn't change however it I can see on the knob screen that it does.  I expect the mesh offset to be aligned with physical coordinates.
* G0 X{something large}.  As [documented](https://marlinfw.org/docs/gcode/G092.html), the physical limits should be maintained however I observe the X carriage crash on the right side of travel.

### Version of Professional Firmware

FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-T13-MPC, based on bugfix-2.1.x (Jan 27 2024 14:31:38) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2-422-BLTUBL-T13-MPC EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

3V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

mriscoc, 2024-06-03:
This behavior can be expected when `NO_WORKSPACE_OFFSETS` is enabled as it is by default in this firmware. This firmware implements variable physical limits as a better approach to keep leveling accuracy.
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-set-the-home-offsets

lagnat, 2024-06-03:
My feeling is that the mesh should respect G92 even when home offsets are disabled.  I'd even go so far as to say that you should disable G92 if your intention is to simplify coordinate calculations however it is my hope that you accept this as a bug and consider determining the level of effort to fix it.

If I build the firmware to re-enable home offsets but set them at 0, 0, would there still be discrepancies to worry about?

Thanks!

closed 2024-08-15 by github-actions[bot]