mriscoc/Ender3V2S1 #1326: Still problems with 300C update

Basmes23, 2024-07-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hello, I read through a post and then downloaded your firmware. I then entered the command C104 U1 T315 and C500 via pronterface. Now I can also reach temperatures of 300C, but after 300C the temperature continues to rise until 312C and then an error message appears. If I follow the temperature in Pronterface, the error appears in the display as soon as the value has risen above 300C. I think I have overlooked or misunderstood something. Can you help me please?
[image]

### Version of Professional Firmware

Ender3V2-422-BLTUBL-T13-MPC-20240127.bin

### Printer model

Ender 3V2

### Electronics

4.2.2

### Other add-ons

2*z Steppermotors, LED light

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2024-07-11:
https://github.com/mriscoc/Special_Configurations/releases/tag/T13

Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2024-07-11:
Maybe related to: https://github.com/MarlinFirmware/Marlin/issues/27253

closed 2024-10-28 by github-actions[bot]