mriscoc/Ender3V2S1 #1328: [BUG] (bug summary)missing files / files renamed?

entityThiex, 2024-07-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I'm trying to compile a custom configuration and have run into a platformio issue, the file 'libproui_ubl_ultipanel.a' cannot be found. I've followed the directions step by step, so this should not happen, right? The file is also not there, I've manually looked it up to make sure it is not a too long path or similiar.

### Expected behavior

Compile without error

### Actual behavior

failed with FileNotFoundError: [Errno 2]

### Steps to Reproduce

follow the steps of your configurator tool, then try to compile it.

### Version of Professional Firmware

Marlin 2.1.1.0

### Printer model

Ender 3 Pro

### Electronics

E3 free-runs TMC2209

### LCD/Controller

12864

### Other add-ons

cr-touch, creality spider hotend

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2024-07-17:
Libraries are here: https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released/Marlin/lib/proui/stm32f1

entityThiex, 2024-07-17:
Thanks, I'll use these

mriscoc, 2024-07-17:
The support for that board is preliminary: https://github.com/MarlinFirmware/Marlin/pull/26562

entityThiex, 2024-07-17:
But wasn't that branch deleted? shouldn't these changes be in the main branch? If so, are they simply not usable in the configurator or unavailable at the moment?

mriscoc, 2024-07-17:
I just check and currently there is no support for the combination: E3 free-runs + Ultipanel. I will try to update the support soon. If you are a Patreon will have access to a pre-release library.

entityThiex, 2024-07-17:
Is there a need to use the ultipanel? I'm using a CR10DSPLY display, is that the problem or am I doing something wrong?

mriscoc, 2024-07-17:
Ultipanel is a generic name for many displays including the CR10DSPLY.

entityThiex, 2024-07-18:
Can you give me some approximate time this will take? You probably have a lot on your plate, so I don't think that it will be fast. Just would be nice to know a roundabout time.

mriscoc, 2024-07-18:
Usually, a previously unsupported combination is compiled when someone requests it on the Patreon site and then the library is made available, also when there are many requests for that specific combination.

entityThiex, 2024-07-18:
What display would I need to have a supported version?

mriscoc, 2024-07-18:
A color display such as DWIN/DACAI/TJC: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display
With those displays you will have additional benefits such as being able to see the leveling grid, a graph of the temperatures, etc.

closed 2024-09-27 by github-actions[bot]