mriscoc/Ender3V2S1 #1333: [BUG] Pause (M25) parks twice

evilDave, 2024-07-23:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

When pausing due to an M25 gcode, the printer parks, then after you hit continue it goes back to the last location and parks again, you hit continue a second time and it will go back to the last location and keep printing from there.

The same does not happen if you hit park from the screen.

similar: https://github.com/mriscoc/Ender3V2S1/issues/743

(like that other issue, I was also having a problem with positioning after the pause - y shift - however that has been fixed by changing the park position to be not right on the edge of the printable area)

### Bug Timeline

now, however I have never pause a print via gcode before

### Expected behavior

Expected to park one time and then resume printing

### Actual behavior

Parked, moved back to location and parked again, then resumed

### Steps to Reproduce

1. Add an M25 command to a gcode file
2. when the pause happens, hit continue
3. the printer will start to resume, then park again
4. hit continue again
5. the printer will actually resume printing

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 v2 neo

### Electronics

422

### LCD/Controller

tjc

### Other add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Additional information & file uploads

Orca Slicer (this could be made an option of the issue template)

evilDave, 2024-07-27:
M600 does not exhibit the same behaviour - but I don't want a filament unload and load in this case.

M25 is what orca slicer inserts if you tell it to add a pause before a layer - perhaps there is a better pause to add? It seem like M25 is supposed to be for an SD card print anyway? I'll try other print commands...

processed-bits, 2024-09-14:
Remove `M25` from Pausing commands in OctoPrint printer settings. The firmware has advanced pause feature, that parks head on pause.

When you have `M25` in your G-code, that triggers firmware parking, and no commands come from OctoPrint until you dismiss/close the pause dialogs. OctoPrint is still waiting to send its pause command (and custom pause scripts), which it does when you continue after the first pause.

You can turn on `serial.log` and see communication with the printer. For example, OctoPrint 'before pausing' custom script runs only before OctoPrint's second pause in such a case. The first pause is handled purely by printer's firmware.

gunderbolt, 2024-11-11:
Thank you so much for mentioning this! That fixed it for me.

Since this isn't a bug with the firmware, I would suggest an update to the Octoprint Wiki Page under the [Protocol Fine Tuning](https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#protocol-fine-tuning) section to remove the M25 gcode from the "Pausing commands" field and replace the image to include both fields that should be edited.

[image]

mriscoc, 2025-01-10:
> I would suggest an update to the Octoprint Wiki Page

Added to the Wiki, Thanks

closed 2025-03-23 by github-actions[bot]