mriscoc/Ender3V2S1 #1334: [FEATURE REQUEST] SD card sorting most recent first

TirunehF, 2024-07-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

In 2022 someone previously raised an issue regarding the option to sort SD card files on the printer screen by "Last Modified" date instead of alphabetically (Issue #232). Although the issue was closed, the feature remains unimplemented. This sorting capability would significantly enhance usability and efficiency, especially when dealing with numerous files. Could you please reconsider this feature or provide guidance on how it might be achieved? The existing advanced features are fantastic, but this small addition would be highly beneficial.

### Bug Timeline

19.07.2022

### Expected behavior

An option to sort files by "Last Modified" would be very convenient.

### Actual behavior

Files are only sorted alphabetically.

### Steps to Reproduce

1. Insert SD card into printer.
2. Navigate to the file list on the printer screen.
3. Observe that files are sorted alphabetically without an option to change sorting criteria.

### Version of Professional Firmware

2.1.3 20240125 - 01:18

### Printer model

Ender 3V2

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2024-07-26:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2024-07-26:
Disable alphabetical sorting: https://marlinfw.org/docs/gcode/M034.html

TirunehF, 2024-07-29:
thanks, will try it out and update whether it worked

closed 2024-10-28 by github-actions[bot]