mriscoc/Ender3V2S1 #1337: [BUG] (bug summary)

dener3id, 2024-08-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I tried to install  "Ender3V2-422-BLTUBL-MPC-20240125.bin" with an 8gb card and now it's stuck at the creality logo. 

### Bug Timeline

Bricked after installing firmware 

### Expected behavior

Installs firmware and runs.

### Actual behavior

Stuck at creality logo

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20240125.bin

### Printer model

3v2

### Electronics

4.2.2

### LCD/Controller

TJC

### Other add-ons

None 

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

None

### Host Software

None

acritum, 2024-08-07:
AFAIR if you install the first time you should update LCD firmware first, and then flash this firmware. Read here: https://github.com/mriscoc/Ender3V2S1/blob/Ender3V2S1-Released/display%20assets/readme.md

dener3id, 2024-08-07:
> AFAIR if you install the first time you should update LCD firmware first, and then flash this firmware. Read here: https://github.com/mriscoc/Ender3V2S1/blob/Ender3V2S1-Released/display%20assets/readme.md

I have a TJC and the instructions are less clear or am I missing something?

acritum, 2024-08-07:
> I have a TJC and the instructions are less clear or am I missing something?

IDK, I have a different LCD and had no issues after updating the display driver. I think you should follow the instructions where it is written "TJC display, currently this display needs a [special compile](https://github.com/mriscoc/Special_Configurations/releases/tag/tjc) to show leveling numeric data in the mesh viewer and some displays need [updated firmware](https://github.com/mriscoc/Ender3V2S1/blob/Ender3V2S1-Released/display%20assets/tjc_update.zip?raw=true) to enable G-code preview and boot properly, more info: https://github.com/mriscoc/Ender3V2S1/issues/542.
TJC"

acritum, 2024-08-07:
Note that the display driver is updated not the same as the printer firmware. You should unscrew the display case and you will find a Micro-SD card slot on the display PCB, use it to update the display driver.

mriscoc, 2024-10-27:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-10-27 by mriscoc