mriscoc/Ender3V2S1 #1346: [BUG] M220 in gcode does not change feedrate

evilDave, 2024-08-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I am printing the speed calibration from teaching tech: https://teachingtechyt.github.io/calibration.html#speed
That calibration uses M220 to increase the feedrate during the test. I see the gcode commands in the gcode file, the speed should be going from 80 to 100, 120, 140, 160 but there are no speed increases during the print.

I can adjust the speed using the LCD panel, which I thought did much the same thing as M220...

### Expected behavior

M220 to change feedrate

### Actual behavior

speed does not change

### Version of Professional Firmware

FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3V2Neo-422-BLTUBL-TJC-LA-MPC-Speaker, based on bugfix-2.1.x (Jun 12 2024 14:51:11)

### Printer model

Ender 3 v2 neo

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

mriscoc, 2024-08-30:
There is not an official precompiled firmware with the name: Ender3V2Neo-422-BLTUBL-TJC-LA-MPC-Speaker in the release page.

pointful-dave, 2024-08-30:
Right you are. It works okay with the release version [Ender3V2-422-BLTUBL-MPC-20240125.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20240122/Ender3V2-422-BLTUBL-MPC-20240125.bin) and the one from the LA branch [Ender3V2-422-BLTUBL-LA-MPC-20240127.bin](https://github.com/mriscoc/Special_Configurations/releases/download/LinearAdvance/Ender3V2-422-BLTUBL-LA-MPC-20240127.bin) so it must be something in my custom configured one... I just used the config tool though... i'll try a few more compiled firmware versions and see if i can narrow down the issue

evilDave, 2024-08-30:
Right you are. It works okay with the release version [Ender3V2-422-BLTUBL-MPC-20240125.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20240122/Ender3V2-422-BLTUBL-MPC-20240125.bin) and the one from the LA branch [Ender3V2-422-BLTUBL-LA-MPC-20240127.bin](https://github.com/mriscoc/Special_Configurations/releases/download/LinearAdvance/Ender3V2-422-BLTUBL-LA-MPC-20240127.bin) so it must be something in my custom configured one... I just used the config tool though... i'll try a few more compiled firmware versions and see if i can narrow down the issue

closed 2024-11-10 by github-actions[bot]