mriscoc/Ender3V2S1 #1348: mainboard fan

Aipoc76, 2024-09-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The fan for the motherboard CR 4.2.7 will only turn on when the part cooling fan turns on.  

### Bug Timeline

Just replaced a 4.2.2 board with a 4.2.7 board and it started.

### Expected behavior

Motherboard fan to run all the time.  (its not going to break any ones electric bill).

### Actual behavior

Only turns on with part cooling fan.  (Not the hot end cooling fan) but the printed part aka extruded material.

### Steps to Reproduce

1. power on system
2. observe mainboard fan not spinning
3. turn on part cooling fan. 
4. turn fan off again and see motherboard fan stop.

### Version of Professional Firmware

Ender3V2-427-BLTUBL-MPC-20240125

### Printer model

Ender 3V2 NEO

### Electronics

4.2.7

### LCD/Controller

Color V2 LCD

### Other add-ons

running a 4020 fan on mainboard.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2024-09-14:
Not a firmware issue. Motherboard fan (K-FAN1) and part cooling fan (K-FAN2) are in parallel.
[image]

Aipoc76, 2024-09-14:
thanks for the info.  Where do you all find these drawings?

mriscoc, 2024-09-15:
> thanks for the info. Where do you all find these drawings?

https://www.reddit.com/r/ender3v2/comments/wglu9u/mainboard_fan_and_cooling_fan/

classicrocker883, 2024-11-03:
i just wire mine with the hotend fan, because thats on all the time

closed 2025-01-13 by github-actions[bot]