mriscoc/Ender3V2S1 #1353: [BUG] (bug summary)Bed tramming error

athandracin, 2024-09-20:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Come up om the screen probe out of bounds after the left front 1st probe.

### Bug Timeline

2 days ago

### Expected behavior

expected to finish tramming.

### Version of Professional Firmware

2.1.3 MRiscoC 20240125-01:21

### Printer model

ender3 v2

### Electronics

4.2.7

### LCD/Controller

not sure

### Other add-ons

cr touch,filament run out

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

orca slicer

athandracin, 2024-09-20:
i think i fixed it.Had to set my z offset in pronterface

GumBall0303, 2024-09-21:
Hi, I just got the firmware installed. And I'm having the same issue with the bed tramming. How did you fix it?

athandracin, 2024-09-22:
Before you start on this procedure you must have the following steps already completed, please read the instructions entirely before starting.

Pronterface installed and working on your computer,

You must be able to connect to your 3D printer and get successful output from the “M503” command.

You must have completed the paper level procedure on your 3D Printer to ensure that the Bed is level as compared to the X gantry

in this article, anything highlighted like this;

image

will have to be typed into the command line interface within pronterface.

G28 - this will home the 3D Printer

G29 - This will run an auto bed level for your 3D Printer

G1 X50 Y30 F1000 - This command will position the nozzle somewhere close to the front of the bed

G1 Z0 - Move nozzle to Z absolute Zero

M114 - This will verify the current position of the Z-Axis

G92 Z10 - This command tells the printer that the Z-axis is at Z=+10 without actually moving the nozzle

M114 - you can verify that the previous command was successful without actually moving the nozzle

M211 S0 Turn Off the Software Endstop

At this point, you can use the Pronterface interface to manually move the Z-Axis/nozzle down or up until a piece of paper just catches between the nozzle and the bed

Once you have the setting correct so the paper is just able to barely move follow the commands below to finish

M114 - You need to see what the 3D printer believes is the new Z Position In the video example below we got back a value of Z:8.5

Now its time for some math

10 - Current Z position + Thickness of paper = new Z offset

(10-8.5+.1= 1.4)

image

Set this new value with the M851 command. (ensure you set it as a negative number)

M851 Z-1.4 - Sets the new Z offset

M503 - Verifies the new value was accepted

M500 Saves the new values to EEPROM

M211 S1 Turn the Software Endstops back on

I have included below 2 links that may be useful as well as providing additional information.

[All3DP – 7 May 22](https://all3dp.com/2/z-offset-3d-printing-how-to-adjust-it/)

[3D Printer Z Offset: Simply Explained 67](https://all3dp.com/2/z-offset-3d-printing-how-to-adjust-it/)
If you're struggling to get your 3D prints to stick or want to print on top of an existing model, Z offset is the answer. Learn more!

[How to Calibrate (Calculate) Z Probe Offset Without LCD over USB with Printrun (Pronterface)](https://www.youtube.com/watch?v=GJiuji4d80M)
How to Calibrate (Calculate) Z Probe Offset Without LCD over USB with Printrun (Pronterface)

athandracin, 2024-09-22:
copied from 3d printing usa forum.

mriscoc, 2024-09-24:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-12-04 by github-actions[bot]