mriscoc/Ender3V2S1 #1356: [BUG] professionalfirmware.py does not work properly on Cura 5.8 on MacOS

RafaelMoreira1180778, 2024-09-27:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The script [slicer scripts/cura/professionalfirmware.py](https://github.com/mriscoc/Ender3V2S1/blob/Ender3V2S1-Released/slicer%20scripts/cura/professionalfirmware.py) is not working properly on Cura 5.8 on MacOS M1.

On Windows 11 Cura 5.8 it works as expected and generates the preview correctly.

### Bug Timeline

2-3 months

### Expected behavior

I expect the script to generate the preview correctly on any platform, being it ARM, AMD64.

### Actual behavior

The script only generates the correct preview on Windows 11.

### Steps to Reproduce

1. Download the script
2. Cura > Help > Open Configuration Folder
3. Copy the script to the `scripts` folder
4. Re-Open Cura
5. Enable the script in post processing
6. Slice a model and generate a preview
7. Insert the SD card on the printer, get the output of the preview:
[image]

### Version of Professional Firmware

Ender3S1-F4-TJC-20230904.bin

### Printer model

Ender3S1 F4

### LCD/Controller

TJC

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

I am running this version so that the mesh view can display the numeric value: https://github.com/mriscoc/Special_Configurations/releases/tag/tjc

mriscoc, 2024-09-30:
Please attach the same file processed on MAC and Windows.

RafaelMoreira1180778, 2024-10-01:
After a bit of debugging I found out that the difference was in the order of script execution. `Professional Firmware Support` needs to run at the end, so at the bottom of the scripts. For some reason another script was overriding the changes made to the `.gcode` file and removing the thumbnail section created by the `professionalfirmware.py` script.

closed 2024-10-01 by RafaelMoreira1180778