mriscoc/Ender3V2S1 #1364: [BUG] (bug summary) losing settings

mips42, 2024-10-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Ender 3 V2, 442 board with Touch sensor, Version 20240122, randomly loses Z offset. Firmware seems to be crashing / resetting for no apparent reason.

### Bug Timeline

Within last week

### Expected behavior

Expected Z offset to remain saved even when machine is off for prolonged periods.

### Actual behavior

randomly loses Z offset. Firmware seems to be crashing / resetting for no apparent reason.

### Steps to Reproduce

So far, there is no set procedure to reproduce the error that I can find.

### Version of Professional Firmware

Ender3v2-422-BLTUBL-MPC

### Printer model

Ender3v2

### Electronics

4.2.2

### LCD/Controller

POLISI3D

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Additional information & file uploads

Currently slicing with Creality print 5.1

mriscoc, 2024-10-15:
I use that version, Ender3V2-422-BLTUBL-MPC-20240125 and don't have this problem.

mips42, 2024-10-15:
okay, Good. I guess. I do. I am now checking Z before every print because I never know when the setting is going to change. While this mitigates the problem, it foes not solve it.

mriscoc, 2024-10-15:
Maybe the motor drivers of your printer don't support the increased speed and acceleration of the new firmware. In that case it is better to return to the stock firmware.

mips42, 2024-10-15:
I suppose its possible but I'm not sure why it would affect only the Z offset. I might switch back but Id miss the viewable mesh.

mriscoc, 2024-10-15:
Each driver is independent, in the case of Z axis during the home and leveling operations the axis changes the direction several times.

mips42, 2024-10-18:
Add-on: after a successful print yesterday, I went to exit back to the main screen and the control knob was non-responsive. After about 5 seconds of this, the system rebooted and the z-offset was reset to zero. I dont know if this helps or not.

mriscoc, 2024-10-18:
> the z-offset was reset to zero

Only the Z-offset was affected or all the settings returned to default?

mips42, 2024-10-18:
Not sure I understand. The Z offset is all that I am changing so, Yes?

mriscoc, 2024-10-27:
Several limits were implemented in Marlin which restrict what values are valid for the probe offsets, so if your current probe offsets values are out of those limits they are considered invalid and they are not taken and saved.  
  
Related to:
https://github.com/MarlinFirmware/Marlin/pull/26774  
https://github.com/MarlinFirmware/Marlin/pull/26267  
https://github.com/MarlinFirmware/Marlin/pull/26397  
https://github.com/MarlinFirmware/Marlin/issues/26417

closed 2024-10-27 by mriscoc