mriscoc/Ender3V2S1 #1368: [BUG] Laser problem with latest FW

kpbicka, 2024-11-04:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The latest firmware version with which the laser works correctly:
**Ender3S1-F4-CVLM-2023**08**12.bin**

with correct execution of G-code
laser settings:
_Ender3S1 with Marlin G-code
BaudRate 250.000
S-value max = 9999
Laser Control Commands = M03/M05_

On latest firmware laser does not work or works incorrectly
firmwares versions with the problem:

Ender3S1-F4-CVLM-2023**09**04.bin
Ender3S1-F4-CVLM-20240127.bin

incorrect execution of G-code - during the execution of G-code the laser does not turn on
laser settings same as earlier:

_Ender3S1 with Marlin G-code
BaudRate 250.000_
_S-value max = 9999_ (any value of this parameter does not eliminate problem)
_Laser Control Commands = M03/M05_ (other _M106/M107_ or **_Inline_** do not fix the problem)

Software used: LightBurn.V1.7.03.dmg for MacOS
Advanced settings - Laser Control - Toggle Laser - Laser ON (Laser diode begin lid on minimal brightness (as usual))

example of G-code generated by LightBurn that is sent to the printer (console out)

Starting stream
G21
G90
M9
M05
G0 X30 Y20 F0
Layer C00
M03 S100
G1 X50F6000
M9
M05
G90
G0 X0 Y0 F0
echo:Unknown command: "M9"
echo:Unknown command: "M9"
Stream completed in 0:00

Laser behavior:
bright flash, move to the starting point of the drawing, burning the drawing (but the laser does not emit power and nothing is drawn), at the end a bright single flash, move to the home position

on the LightBurn forum (https://forum.lightburnsoftware.com/t/ender-3-s1-pro-tips-tricks-and-suggestions/141611) they advise using
Start G-code command = M3
This advice does not completely help
Here is the laser behavior:
bright flash, move to the starting point of the drawing, burning the drawing at an unknown power (the laser emits power, but the program sets the value to 1%), at the end a bright single flash, move to the home position

Here is the G-code

Starting stream
M3
G21
G90
M9
M05
G0 X30 Y20 F0
Layer C00
M03 S1000
G1 X50F6000
M9
M05
G90
G0 X0 Y0 F0
echo:Unknown command: "M9"
echo:Unknown command: "M9"
Stream completed in 0:00

Expected laser behavior - moving to the starting point, burning at a given power, turning off the laser. Unfortunately, no settings can achieve success.

Previously, on the Ender3S1-F4-CVLM-20230812.bin firmware, no settings were required and the laser worked correctly.
If no change settings in LightBurn and downgrade mriscoc-firmware to Ender3S1-F4-CVLM-2023**08**12.bin then laser works fine!

### Version of Professional Firmware

FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3S1-F4-CVLM, based on bugfix-2.1.x (Jan 27 2024 19:26:39)

### Printer model

Creality Ender3S1 F4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2024-12-02:
Something was changed in the Marlin code base that maybe broke the laser functionality, unfortunately I don't have currently the hardware to do the test to solve this issue.

kpbicka, 2025-04-07:
ясно

TheWebMachine, 2025-05-20:
Unfortunately, I can't help test this, either. I have neither a Mac nor have I ever successfully tried streaming GCode (Octoprint or otherwise) for laser to the E3. Since so much manual adjustment is often needed to setup each piece to be engraved/cut, I figured there was little point in being able to control/send gcode remotely. I've stuck with SD card for all laser gcode and left connected/streaming control (Octoprint in my case) only for FDM gcode.

mriscoc, 2025-07-20:
@TheWebMachine 

Hi, did you test the latest versions? the OP indicated a change in the behavior of the current versions versus the 20230812, not related to remote gcode but a local SD file.