mriscoc/Ender3V2S1 #1381: [BUG] M503 C doesn't save to the SD Card

amandamana, 2024-12-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I can't get M503 C to work. I've tried by printing a G-code configuration file and by issuing the command over a serial connection. Perhaps the required code was disabled to make space in the firmware?

### Expected behavior

I understand the command should write the runtime-configurable settings (in SRAM) out to the SD Card as 'mc.zip'

### Actual behavior

No file is created on the SD card, even after waiting multiple minutes. M503 without the C parameter does work for me over serial connection.

### Version of Professional Firmware

Ender3V2-427-MM-MPC-20240125

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2024-12-10:
Not a firmware issue, you need to enable `CONFIGURATION_EMBEDDING` for be able to use the [C] parameter: https://marlinfw.org/docs/gcode/M503.html

amandamana, 2024-12-10:
Any pros or cons to enabling this on the precompiled releases?

closed 2024-12-10 by amandamana