mriscoc/Ender3V2S1 #1382: Ender3V2NEO-bed level issue

Ruliman830, 2024-12-10:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have installed the latest firmware. have done the tramming feature and bed is with I specs have gotten it as close as .1 across. Then I do the auto mesh and when I do a test print it prints low on the far left side than on the far right side. Have tried to edit the mesh but don’t see any changes at all. Anybody has had this issue before? 
[image]
[image]

### Expected behavior

I am unsure if red numbers are close to nozzle or far? 

### Version of Professional Firmware

Version 2.1.3 MRiscoC 20240125

### Printer model

Ender 3V2NEO

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

mriscoc, 2024-12-10:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

Ruliman830, 2024-12-11:
Thank you for the info and sorry for this post I am new to this GitHub.

closed 2024-12-11 by Ruliman830