mriscoc/Ender3V2S1 #1388: [BUG] TYPO in AML documentation

HankLumbard, 2024-12-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

; Heat up
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
G28 ; Home all axes
G27 ; Park tool head
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
;
; Reset settings
M220 S100 ;Reset Feed rate
M221 S100 ;Reset Flow rate
;
; Ender Custom Start G-code
G92 E0 ; Reset Extruder
**_G28O ; Home optionally if steppers were shutdown_**

This line should read G28 0 .....

M420 S1 ; activate leveling
;
; Adaptive Mesh Leveling (AML)
C29 A ; 

### Bug Timeline

Just noticed it

### Expected behavior

my printed beeped at me

### Actual behavior

it's just a typo

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-LA-MPC, based on bugfix-2.1.x (Dec 15 2023 12:57:55)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2024-12-17:
Be sure that you write G28 O (letter O, not a number Zero)

HankLumbard, 2024-12-17:
Well how about that? A typo in my typo post! Thank you. SO am I wrong then? Do you need a space after the G28 at all? Feel free to delete this whole thread if so.

closed 2025-02-27 by github-actions[bot]