mriscoc/Ender3V2S1 #1394: Have upgraded to 20240122 version and get 'Printer Killed - Homing Failed' (Mainboard is 4.2.7 with GD32 soc)

Warfare360, 2025-01-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Have upgraded to 20240122 version and get 'Printer Killed - Homing Failed' (Mainboard is 4.2.7 with GD32 soc).
My board is using GD32 soc, version is 4.2.7.
Previous release was working fine : Ender3V2-422-BLTUBL-MPC-20231217.bin

### Bug Timeline

Started with 20240122 version

### Expected behavior

Homing functionning well

### Actual behavior

Printed is stucked displaying " Printer Killed - Homing Failed ".
Impossible to use it.

### Steps to Reproduce

Power on.
Home the printer.
=> Fail

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20240122.bin

### Printer model

Creality Ender 3 V2

### Electronics

Board 4.2.7

### LCD/Controller

Original

### Other add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

Pdub9, 2025-01-05:
The same thing is happening to me.

mriscoc, 2025-01-05:
Hi, this is not a general printer support forum. Many problems can be easily solved by reading the firmware [Wiki](https://github.com/mriscoc/Ender3V2S1/wiki#main-topics) please check the calibration guide.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2025-01-08 by Warfare360