mriscoc/Ender3V2S1 #1398: [BUG] octoprint keeps loosing serial connection to new experimental build

otkaz, 2025-01-12:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I just upgraded to the new experimental build from 20240122 on my stock ender3s1 f4. Used this file
https://github.com/mriscoc/Ender3V2S1/releases/download/20250106/Experimental-Ender3S1-F4-UBL-20250106.bin
I have it connected to octopi and I keep getting comm issues and have to reconnect after about 5 mins or so. I tried switching the 115k baud but the issue persists. Switched back to the old build and issue went away. I didn't see anything in the octoprint logs except for a comm error that it lost connection.

### Bug Timeline

New issue with experimental

### Steps to Reproduce

Just connect to octoprint and wait. After a while the comm port will time out with an error in octopi about loosing serial connection. It reconnects when I hit the connect button but will eventually disconnect again

### Version of Professional Firmware

Experimental-Ender3S1-F4-UBL-20250106

### Printer model

S1 F4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

otkaz, 2025-02-12:
I'm still having this issue any time I try the experimental firmware with octoprint. Is anyone else running into this or is it a problem with just my setup? I don't have this issue on the sable build.

`No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor`

bbrowning918, 2025-03-27:
I had a similar sounding issue with timeouts on (newer) builds of this firmware and I found [this post from the octoprint forums](https://community.octoprint.org/t/too-many-consecutive-timeouts-printer-still-connected-and-alive/36469/2) to be very helpful. A new/different cable with the ferrite beads made everything work again.

My previous/repurposed from I don't even remember cable worked on older (Marlin) builds of all kinds, and I had not any problems _previously_. My only guess is something in the latest experimental perhaps pushes an otherwise barely usable cable to cause issues.

closed 2025-06-06 by github-actions[bot]