mriscoc/Ender3V2S1 #1406: Unable to compile firmware on VS studio code

slholt, 2025-02-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

 "DEFAULT_EJERK" is undefined [Ln 949,Col 45] 'DEFAULT_EJERK' was not declared in this scope [Ln949, Col 70]

### Bug Timeline

NEW  (02/08/20250

### Expected behavior

I was trying to compile for 4.2.7 board for cr10spro 

### Actual behavior

Would not compile Error

### Version of Professional Firmware

Newest

### Printer model

CR10SPRO V2

### Electronics

4.2.7

### LCD/Controller

Ender V2 Screen or Bigtreetech tft43

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Same as my slicer

### Additional information & file uploads

Was following a walk through video by Victor Bared, Was going to switch out my stock creality board to 4.2.7. Also was going to run clipper on pi and change stock screen to one of the above.

midaryl, 2025-02-11:
I have MRiscoc 2.1.3 precompliled firmware for a CR10S Pro V2

slholt, 2025-02-14:
> I also have the same error. Compiling using the special configurator. settings Ender3V2Neo, 4.2.7, BLT, DWIN, T13 and IS, LA, MPC, Speaker. I need to make a new version the printer tells me it's out of bounds so it will not do anything. This is without me changing anything. (I now have the experimental version of the firmware, so I thought reverting back to the regular one would eliminate the problem.) Any suggestions on the compiling error or the out-of-bounds error?
>
> I also have the same error. Compiling using the special configurator. settings Ender3V2Neo, 4.2.7, BLT, DWIN, T13 and IS, LA, MPC, Speaker. I need to make a new version the printer tells me it's out of bounds so it will not do anything. This is without me changing anything. (I now have the experimental version of the firmware, so I thought reverting back to the regular one would eliminate the problem.) Any suggestions on the compiling error or the out-of-bounds error?

I did figure this out. You have to use the experimental version. I was able to finally compile this firmware. I have not installed yet but I have a bin. 

Yahoo Mail: Search, Organize, Conquer

melyux, 2025-04-17:
I'm on the experimental version, still getting this error.

RoyalHawk2002, 2025-05-25:
If I try to compile the FW "Ender3V2-427-BLTUBL-T13-LA-MPC" in VSC , but I got the following wrrors when I use the files generated by the Configurators, I did a try with the original source code "Ender3V2S1-20240122" and it works. I want to add LA and support for Creality Cloud.

In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:273:11: error: missing binary operator before token "(" 
  273 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:972:11: error: missing binary operator before token "(" 
  972 | #if EITHER(MORGAN_SCARA, MP_SCARA)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1823:11: error: missing binary operator before token "(" 1823 | #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:2060:11: error: missing binary operator before token "(" 2060 | #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3490:11: error: missing binary operator before token "(" 3490 | #if EITHER(RGB_LED, RGBW_LED)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:56,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1181:11: error: missing binary operator before token "("
 1181 | #if EITHER(INPUT_SHAPING_X, INPUT_SHAPING_Y)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1423:11: error: missing binary operator before token "("
 1423 | #if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1481:28: error: missing binary operator before token "("
 1481 | #if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
      |                            ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1551:15: error: missing binary operator before token "("
 1551 |     #if EITHER(HAS_MARLINUI_U8GLIB, TFT_COLOR_UI)
      |               ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1557:15: error: missing binary operator before token "("
 1557 |     #if EITHER(HAS_MARLINUI_U8GLIB, TOUCH_UI_FTDI_EVE)
      |               ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1622:11: error: missing binary operator before token "("
 1622 |   #if BOTH(M73_REPORT, HAS_MEDIA)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1628:26: error: missing binary operator before token "("
 1628 | #if HAS_DISPLAY && EITHER(HAS_MEDIA, SET_PROGRESS_MANUALLY)
      |                          ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2308:11: error: missing binary operator before token "("
 2308 | #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2316:9: error: missing binary operator before token 
"("
 2316 | #if BOTH(AUTO_BED_LEVELING_UBL, EEPROM_SETTINGS)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2428:11: error: missing binary operator before token "("
 2428 | #if EITHER(ARC_SUPPORT, BEZIER_CURVE_SUPPORT)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2514:9: error: missing binary operator before token 
"("
 2514 | #if BOTH(HAS_MEDIA, DIRECT_STEPPING)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:3459:11: error: missing binary operator before token "("
 3459 | #if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
      |           ^
 "("
 4211 | #if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
      |           ^
Error: Failed to parse Marlin features. See previous error messages.

mriscoc, 2025-05-26:
> I got the following errors when I use the files generated by the Configurators

Hi, be sure that you are using compatible versions of the repository, currently there are two versions: 2024 y 2025:

** For stable 2024 use:
https://github.com/mriscoc/Special_Configurations/tree/New-Year-2024-compatible-version
https://github.com/mriscoc/Ender3V2S1/tree/New-Year-2024-compatible-version

** For experimental 2025 use:
https://github.com/mriscoc/Special_Configurations/tree/main
https://github.com/mriscoc/Ender3V2S1/tree/Experimental-20250111

Kenyiro34, 2025-05-30:
I got same problem , I found my Old special configuration on anorher hdd and when i create new ender3v2 ubl bltouch skr mini e3v2 cr10display and home,ia,la,mpc,  and click make not have errors and copy conf.adv h to marlin bugfix 2.1.x and try make firmware always got error and firmware fail. Older version special configuration got same errors , I think change some on vsc new release. But if i use conf.h adv from folder configuration examples , firmware success compiling. Maybe on special configuration change something thats always we got fail compiling.