mriscoc/Ender3V2S1 #1410: [BUG] (bug summary)

Slavikpd, 2025-02-12:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hello. I want to connect the Ender 3V2 Creality Wifi Box 2.0 to the printer. I have the Wifi box installed (LAN), but it does not recognize the printer. The printer firmware is mriscoc Ender3V2-422-BLTUBL-MPC-20240125. Do you have experience with this? Thank you.

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20240125

### Printer model

Creality Ender 3V2

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

Koky05, 2025-02-13:
It is mentioned in Wiki that for work with Creality WiFi box you need custom FW with enabled support of Creality Cloud. Also WiFi box expect communication on other Baudrate (115200) as default FW provide (250000), but that can be change in Advance Menu bet activating 115K baud.

[image]

Slavikpd, 2025-02-13:
Thank you for your support. Everything is going well. I consider it
resolved.

closed 2025-04-25 by github-actions[bot]