mriscoc/Ender3V2S1 #1411: [BUG] The steppers are too hot -

Czarnyszarf, 2025-02-12:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The steppers on printer are getting too hot when using newest 4.2.2 firmware (Ender3V2-422-BLTUBL-MPC-20240125)

They are painful to the touch, much hotter than on stock firmware. 

All stepper motors are hot. 

### Expected behavior

Decrease RMS/ Voltage or enable gcode to modify it

### Actual behavior

Stepper being cooler - right now I would be afraid to run the printer in the hot summer, the extruder for example could get too hot and deform printed PETG parts. 

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20240125

### Printer model

Ender 3v2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Egosumumbravir, 2025-03-13:
You can't use software to change the VREF on the stock Creality boards.  It's simply not supported.
The only way is by adjusting the potentiometers on the board - https://www.youtube.com/watch?v=k2nm-WetPnU

If you want this kind of function, you';ll need to upgrade to a board with UART control of the drivers.

mriscoc, 2025-03-14:
Lower the maximum Axes speed

closed 2025-05-24 by github-actions[bot]