mriscoc/Ender3V2S1 #1416: undefined reference to `Buzzer::tone

Lazox12, 2025-02-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I can`t compile the ubl firmware for the manta M5P board. the bilinear version works 

### Bug Timeline

started today

### Expected behavior

for it to compile :D

### Actual behavior

It didn`t compile :D error log : 
```
Linking .pio/build/STM32G0B1RE_manta_btt/firmware.elf
/home/lazox/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: Marlin/lib/proui/libproui.a(proui_ex.cpp.o): in function `MarlinSettings::load_mesh(signed char, void*)':
(.text._ZN14MarlinSettings9load_meshEaPv+0x7c): undefined reference to `Buzzer::tone(unsigned short, unsigned short)'
/home/lazox/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: (.text._ZN14MarlinSettings9load_meshEaPv+0x84): undefined reference to `Buzzer::tone(unsigned short, unsigned short)'
/home/lazox/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: (.text._ZN14MarlinSettings9load_meshEaPv+0x8c): undefined reference to `Buzzer::tone(unsigned short, unsigned short)'
/home/lazox/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld: (.text._ZN14MarlinSettings9load_meshEaPv+0xb2): undefined reference to `Buzzer::tone(unsigned short, unsigned short)'
```

### Steps to Reproduce

building the code with pio i`ve used: ~/.platformio/penv/bin/pio run -e STM32G0B1RE_manta_btt

### Version of Professional Firmware

newest today`s clone of the repo

### Printer model

Ender3 with bl touch and 

### Electronics

Board: BigTreeTech Manta M5P

### LCD/Controller

the display from ender 3V2 with the knob

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.h.txt](https://github.com/user-attachments/files/18874263/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/user-attachments/files/18874262/Configuration_adv.h.txt)
other source code on this repo : https://github.com/Lazox12/ender

pls help
I`m sory for my english

mriscoc, 2025-02-21:
The Manta M5P board is not in the list of supported boards.

Lazox12, 2025-02-27:
yeah I know Im just looking if someone has got this bug before or if its just me or maybe for some potential fixes

mriscoc, 2025-02-27:
It is not a firmware bug, you can open a discussion here: https://github.com/mriscoc/Ender3V2S1/discussions

closed 2025-02-27 by mriscoc